TheDeanLab/navigate #434: Pause during confocal projection mode

BingyingChen, 2023-04-25:
```
Waveform with 1.0 planes is of length (22500,)
c:\users\multiscale\git\aslm\src\aslm\controller\sub_controllers\waveform_tab_controller.py:288: UserWarning: Creating legend with loc="best" can be slow with large amounts of data.
  self.view.fig.tight_layout()
C:\Users\MultiScale\.conda\envs\ASLM\lib\tkinter\__init__.py:814: UserWarning: Creating legend with loc="best" can be slow with large amounts of data.
  func(*args)
C:\Users\MultiScale\.conda\envs\ASLM\lib\site-packages\nidaqmx\errors.py:140: DaqWarning:
Warning 200010 occurred.

Finite acquisition or generation has been stopped before the requested number of samples were acquired or generated.
  warnings.warn(DaqWarning(
C:\Users\MultiScale\.conda\envs\ASLM\lib\site-packages\nidaqmx\errors.py:140: DaqWarning:
Warning 200010 occurred.

Finite acquisition or generation has been stopped before the requested number of samples were acquired or generated.
  warnings.warn(DaqWarning(
```

[image]

In the developed branch, there's a pause during the acquisition under confocal projection mode. The settings and error message shows above. 

[model_debug.log](https://github.com/TheDeanLab/ASLM/files/11326019/model_debug.log)
[view_controller_debug.log](https://github.com/TheDeanLab/ASLM/files/11326021/view_controller_debug.log)

zacsimile, 2023-04-26:
One thing to note is here is `c:\users\multiscale\git\aslm\src\aslm\controller\sub_controllers\waveform_tab_controller.py:288: UserWarning: Creating legend with loc="best" can be slow with large amounts of data.
  self.view.fig.tight_layout()` runs even when the camera tab is selected. This indicates that the waveform controller plotting is getting executed even when the plot is not visible. This is a new thing, and is rather expensive for confocal projection mode. We should switch it back to only plotting when the waveform tab is accessed.

BingyingChen, 2023-05-03:
Annie found out this is due to the length difference between the wavefront and the camera. The error in the green channel has been fixed. But it remains in the rest channels and probably causes another issue

[image]

closed 2023-05-08 by zacsimile