TheDeanLab/navigate #438: Associate metadata with MIPs

zacsimile, 2023-05-02:
We'd like to store the full dataset's metadata in the header of its MIP. I propose moving the MIP saving to the data source classes to most easily accomplish this. Any objections, @AdvancedImagingUTSW? @annie-xd-wang?

AdvancedImagingUTSW, 2023-05-02:
Seems very reasonable as a form of redundancy

zacsimile, 2023-05-02:
I'm particularly interested in doing this for the OMERO import.

AdvancedImagingUTSW, 2023-05-05:
Would be nice to know what git hash data was collected with too.

[for example...](https://stackoverflow.com/questions/14989858/get-the-current-git-hash-in-a-python-script)

zacsimile, 2023-05-05:
We presently stash the ASLM software version in the metadata. We could make it more specific with a hash. Let's make a different issue for this, since it's not MIP-specific.

zacsimile, 2023-05-05:
Presently inclined to not add metadata to MIPs, as it may not be necessary. Worked with BioHPC yesterday and we discovered that the data transfer rates for symlinks to full data sets was sufficiently low to be teneble. New plan is to create a symlink to full data sets stored in `/archive/shared/MIL` and then `omero import` the symlink.

zacsimile, 2023-05-19:
No longer necessary.

closed 2023-05-19 by zacsimile