TheDeanLab/navigate #444: ROI setting doesn't work for M1 Mac in synthetic hardware mode

zacsimile, 2023-05-10:
It appears the SharedNDArray memory isn't released.

```
2023-05-09 15:49:51,516 - controller - ERROR - camera_setting_controller: could not convert string to float: '' similar to TclError
2023-05-09 15:49:53,498 - controller - ERROR - camera_setting_controller: could not convert string to float: '' similar to TclError
Traceback (most recent call last):
  File "/Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/multiprocessing/resource_tracker.py", line 201, in main
    cache[rtype].remove(name)
KeyError: '/psm_962534db'
Exception inside ObjectInSubprocess: Traceback (most recent call last):
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/concurrency/concurrency_tools.py", line 93, in __new__
    shm = shared_memory.SharedMemory(create=True, size=requested_bytes)
  File "/Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/multiprocessing/shared_memory.py", line 114, in __init__
    self._mmap = mmap.mmap(self._fd, size)
OSError: [Errno 24] Too many open files

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/concurrency/concurrency_tools.py", line 521, in _child_loop
    result = getattr(obj, method_name)(*args, **kwargs)
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/model.py", line 312, in get_data_buffer
    self.update_data_buffer(img_width, img_height)
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/model.py", line 279, in update_data_buffer
    self.data_buffer = [
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/model.py", line 280, in <listcomp>
    SharedNDArray(shape=(img_height, img_width), dtype="uint16")
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/concurrency/concurrency_tools.py", line 96, in __new__
    raise OSError(
OSError: You tried to simultaneously open more SharedNDArrays than are allowed by your system!

Exception in Tkinter callback
Traceback (most recent call last):
  File "/Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
  File "/Users/zachmarin/Code/ASLM/src/aslm/controller/sub_controllers/acquire_bar_controller.py", line 296, in launch_popup_window
    self.parent_controller.execute("acquire")
  File "/Users/zachmarin/Code/ASLM/src/aslm/controller/controller.py", line 953, in execute
    if not self.prepare_acquire_data():
  File "/Users/zachmarin/Code/ASLM/src/aslm/controller/controller.py", line 732, in prepare_acquire_data
    self.update_buffer()
  File "/Users/zachmarin/Code/ASLM/src/aslm/controller/controller.py", line 276, in update_buffer
    self.data_buffer = self.model.get_data_buffer(img_width, img_height)
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/concurrency/concurrency_tools.py", line 412, in attr
    return _get_response(self, True)
  File "/Users/zachmarin/Code/ASLM/src/aslm/model/concurrency/concurrency_tools.py", line 433, in _get_response
    raise resp
Exception: You tried to simultaneously open more SharedNDArrays than are allowed by your system!
```

AdvancedImagingUTSW, 2023-05-22:
I also was able to replicate this on my computer, and found another bug when doing so.  When I try to change the ROI size, I get the string literal empty error that results from us not using a validated spin box in camera_tab.py. 

The number of pixels is also hard-coded, when it should be pulled from the configuration instance (which itself should not be hardcoded in the configuration.yaml file, but rather pulled from the camera).

```
            # FOV Frame
            self.inputs[self.fov_names[i]] = LabelInput(
                parent=self.fov_frame,
                label=self.xy_labels[i],
                input_class=ttk.Entry,
                input_var=tk.IntVar(),
                input_args={"width": 7},
            )
            self.inputs[self.fov_names[i]].grid(row=i, column=0, pady=1, padx=5)

            # ROI Center
            self.inputs[self.center_names[i]] = LabelInput(
                parent=self.center_roi_frame,
                label=self.xy_labels[i],
                input_class=ttk.Spinbox,
                input_var=tk.IntVar(),
                input_args={"from_": 0, "to": 2048, "increment": 1.0, "width": 6},
            )
```

AdvancedImagingUTSW, 2023-05-24:
Was glancing at this. 

Why do we define an array of 100 SharedNDArray objects in the update_data_buffer of the mode like this?
```
        self.data_buffer = [
            SharedNDArray(shape=(img_height, img_width), dtype="uint16")
            for i in range(self.number_of_frames)
        ]
```

This seems to be the problem, at least on my computer. However, making changes to it also results in other issues...

zacsimile, 2023-05-24:
I think the problem is the previously-allocated `SharedNDArray`s are not released on Mac when we update the ROI size. However, the fact that there is a maximum suggests we will reach a cap on buffer size with this method of defining the `data_buffer`. We could instead move to a 3D `SharedNDArray` of size `(buffer_size, img_height, img_width)`, which would solve this issue. Hamamatsu does support this style of contiguous buffer.

zacsimile, 2023-09-20:
This appears to be fixed by `ulimit -n 4096`. I think it just takes a bit for garbage collection to catch up. Expanding the file limit solves our issues locally.

closed 2023-09-20 by zacsimile