TheDeanLab/navigate #445: Add ability to set arbitrary ROI

zacsimile, 2023-05-10:
By default, we cut about the middle of the FOV. However, it is perfectly reasonable to create an ROI from, say, `(500, 430)` to `(700, 490)` or something similarly random. We should implement this in the logic/GUI rather than set the width/height and ROI center (which is equivalent, but currently disabled).

AdvancedImagingUTSW, 2023-11-13:
Assigning @Johnhaug223 to this as his first task. Should be decently localized, and therefore not too challenging. Any things to watch out for would be valuable. The only thing that comes to mind is that different cameras have different step sizes (e.g., the interval at which you can adjust a particular setting), in this case width and height.

zacsimile, 2023-11-13:
We grab these step sizes whe we set up the camera. They are under the attributes `step_image_width` and `step_image_height` in the `HamamatsuOrca` camera class.