TheDeanLab/navigate #447: Some odd default values experienced

AdvancedImagingUTSW, 2023-05-11:
While setting up CT-ASLM-V1 to run on this software, I noticed a few somewhat odd pieces of behavior. Nothing major, and I think I know the source of it.  

Let's say I change the filter names in the configuration file. Well, when I open the software, it uses the experiment file to load the default settings in the view, and now, these don't match what is in the configuration. It almost seems like we should overwrite the experiment file in the .config folder every time we close the software.  This way, when you open it, it will load what you used last time, not some odd default behavior that may not be relevant...

AdvancedImagingUTSW, 2023-05-17:
@annie-xd-wang, @zacsimile. 

This branch is a bit confusing since I made a mistake with git. Essentially, I updated the controller so that it automatically saves over the experiment.yml file upon closing the file. All fine, Zach also agrees that this seems like a good idea.  

However, after I did that, I merged it into another branch where I was working on the ASI filter wheels. As a result of a mistake, all of those changes are now in this branch...  Anyhow, we will just make a single pull request, and we are getting close to being ready.  But, there is one thing that I wanted to discuss. 

We have a few instances where multiple devices share a common API/communication method. For example, the NIGalvoStage, lasers, and shutters, all use the NIDAQ device. Consequently, we have to pass a single `device_connection` to each of these in the `microscope` object. Currently, most of these are kinda hard-coded in.

Would it make more sense to complicate our configuration.yml file a bit more to distinguish hardware that is shared versus those that are not?  For example, something like this?

```
shared_hardware:
  -
      type: ASIFilterWheel and ASIStage
      port: COM19
      baudrate: 9600
  -
      type: NI and NILaser and NIGalvo
      port: ...
      baudrate: ...
hardware:
  daq:
    type: NI
  camera:
    -
      type: HamamatsuOrca
      serial_number: 302158
    -
      type: HamamatsuOrca
      serial_number: 302352
  filter_wheel:
    type: SutterFilterWheel
    port: COM19
    baudrate: 9600
    number_of_wheels: 2
```

This could make the microscope object a little less kludged?

zacsimile, 2023-05-17:
Sorry, it's been a minute since I looked deep into this. Can you provide an example or two of the kludges this would avoid?

AdvancedImagingUTSW, 2023-05-17:
In `microscope.py`

line 179:
```
                elif device_ref_name.startswith("NI") and (
                    device_name == "galvo" or device_name == "remote_focus_device"
                ):
                    # TODO: Remove this. We should not have this hardcoded.
                    device_connection = self.daq
```

line 233:
```
            if device_ref_name.startswith("GalvoNIStage"):
                # TODO: Remove this. We should not have this hardcoded.
                devices_dict["stages"][device_ref_name] = self.daq
```

I also assume there is a good possibility that I am just being a bit dense.

zacsimile, 2023-05-17:
I'll have to go through it in detail but there's probably a way to get the `daq` to be passed on these devices on startup without needing additional specification.

AdvancedImagingUTSW, 2023-05-18:
I'm a hot mess, and somehow messed up my branch. I had to pull everything to a new branch...

closed 2023-05-22 by AdvancedImagingUTSW