TheDeanLab/navigate #450: Large number of entries in a multiacquisition table slows down initial imaging

zacsimile, 2023-05-18:
After running `VolumeSearch`, if I have a large number of entries in the multi-acquisition table, it takes a long time to start an acquisition. I press `Acquire` and it takes about 30 seconds to start. If I delete the positions, it is quick again.

AdvancedImagingUTSW, 2023-05-20:
We naively jammed the pandas table into there since it worked with tk, and Annie kindly made it functional. But it may honestly be one of our weakest dependencies. We could easily consider something else.

zacsimile, 2023-11-16:
Further investigation reveals the slowdown comes not from `pandas`, but from `aslm.config.config.build_nested_dict()`. Likely from multiple assignments to the `multiprocessing.Manager()` object, possibly slower due to recursion.

AdvancedImagingUTSW, 2024-05-21:
Now that I am using the Volume Search feature, this is rearing its ugly head again...

AdvancedImagingUTSW, 2024-05-21:
I have also seen something where the multi position table must have at least one entry in it, or else an error is thrown.

closed 2024-05-23 by AdvancedImagingUTSW