TheDeanLab/navigate #473: ZStack `per_stack` and `per_z` return different channel orderings

zacsimile, 2023-06-03:
Strangely, `per_z` and `per_stack` seem to have different channel orders. For the same configuration in the GUI, Channel 2 on `per_stack` may be Channel 0 or 1 when `per_z` is selected.

[image]

[image]

AdvancedImagingUTSW, 2023-07-10:
@zacsimile - Annie is wondering if this is solved. If so, perhaps she can write a few unit tests to lock in the behavior.

zacsimile, 2023-07-10:
It is not solved.

zacsimile, 2023-07-21:
For example, on the v2, per-stack saves in the correct order of (channel 0, channel 1, channel 2), but per z saves in the order of (channel 1, channel 2, channel 0). This does appear to be a saving issue, rather than an acquisition order issue.

Additionally, in per-stack (but not per-z), the final frame of the z-stack is in the first position of the stack. This is starting to feel like an issue with `_cztp_indices()`.

[image]

closed 2023-07-23 by zacsimile