TheDeanLab/navigate #480: Add graceful close to image data if `OSError` is hit

zacsimile, 2023-06-09:
Call `self.close()` in data source if we reach

```
Traceback (most recent call last):                                                                                                                                                                                                             File "h5py\_objects.pyx", line 201, in h5py._objects.ObjectID.__dealloc__                                                                                                                                                                  OSError: [Errno 28] Driver write request failed (file write failed: time = Thu Jun  8 16:46:45 2023                                                                                                                                          , filename = 'C:\Users\DeanLab\Desktop\Data\Zach\Lung\AB7389\antiGFP-AF488_TOPRO647\2023-06-05\Cell_010\z-stack\CH00_000000.hdf', file descriptor = 9, errno = 28, error message = 'No space left on device', buf = 00000194828CF1F8, total write size = 8192, bytes this sub-write = 8192, bytes actually written = 18446744073709551615, offset = 6018772578008)                                                                                                                        Exception ignored in: 'h5py._objects.ObjectID.__dealloc__'                                                                                                                                                                                   Traceback (most recent call last):                                                                                                                                                                                                             File "h5py\_objects.pyx", line 201, in h5py._objects.ObjectID.__dealloc__                                                                                                                                                                  OSError: [Errno 28] Driver write request failed (file write failed: time = Thu Jun  8 16:46:45 2023                                                                                                                                          , filename = 'C:\Users\DeanLab\Desktop\Data\Zach\Lung\AB7389\antiGFP-AF488_TOPRO647\2023-06-05\Cell_010\z-stack\CH00_000000.hdf', file descriptor = 9, errno = 28, error message = 'No space left on device', buf = 00000194828CF1F8, total write size = 8192, bytes this sub-write = 8192, bytes actually written = 18446744073709551615, offset = 6018772578008)  
```

in `write()`. Test that this works.

AdvancedImagingUTSW, 2023-07-12:
Glancing at this now... Created a partition on my hard-drive with 1GB of space for testing.

for a tiff:
```
Traceback (most recent call last):
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/model.py", line 733, in run_data_process
    data_func(frame_ids)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/features/image_writer.py", line 190, in save_image
    self.data_source.write(
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/data_sources/tiff_data_source.py", line 151, in write
    self.image[c].write(
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/tifffile/tifffile.py", line 1633, in write
    fh.write_array(data)
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/tifffile/tifffile.py", line 10648, in write_array
    self._fh.write(data.tobytes())
OSError: [Errno 28] No space left on device
```

For OME-Tiff
```
Mesoscale: stage axes mapping is not specified in the configuration file, will use the default one in the code!
Nanoscale: stage axes mapping is not specified in the configuration file, will use the default one in the code!
Nanoscale: stage axes mapping is not specified in the configuration file, will use the default one in the code!
Exception in thread z-stack Data:
Traceback (most recent call last):
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/model.py", line 758, in run_data_process
    self.end_acquisition()  # Need this to turn off the lasers/close the shutters
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/model.py", line 678, in end_acquisition
    self.image_writer.close()
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/features/image_writer.py", line 295, in close
    self.data_source.close()
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/data_sources/tiff_data_source.py", line 244, in close
    self.metadata.to_xml(
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/metadata_sources/ome_tiff_metadata.py", line 200, in to_xml
    return super().to_xml(file_type, root=root, **kw)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/metadata_sources/metadata.py", line 233, in to_xml
    d = getattr(
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/metadata_sources/ome_tiff_metadata.py", line 180, in ome_tiff_xml_dict
    "PositionX": views[view_idx]["x"],
IndexError: list index out of range
Exception in thread z-stack Data:
Traceback (most recent call last):
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/tifffile/tifffile.py", line 10645, in write_array
    numpy.ascontiguousarray(data).tofile(self._fh)
OSError: 4194304 requested and 0 written

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/model.py", line 733, in run_data_process
    data_func(frame_ids)
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/features/image_writer.py", line 190, in save_image
    self.data_source.write(
  File "/Users/S155475/Desktop/GitHub/ASLM/src/aslm/model/data_sources/tiff_data_source.py", line 147, in write
    self.image[c].write(data, description=ome_xml, contiguous=True)
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/tifffile/tifffile.py", line 2668, in write
    fh.write_array(data)
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/tifffile/tifffile.py", line 10648, in write_array
[... 22 lines cut ...]
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/dataset.py", line 982, in __setitem__
    self.id.write(mspace, fspace, val, mtype, dxpl=self._dxpl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5d.pyx", line 232, in h5py.h5d.DatasetID.write
  File "h5py/_proxy.pyx", line 114, in h5py._proxy.dset_rw
OSError: [Errno 28] Can't write data (file write failed: time = Tue Jul 11 19:55:37 2023
, filename = '/Volumes/test/Kevin/Lungww/MV3ww/GFPww/2023-07-11/Cell_004/CH00_000000.h5', file descriptor = 221, errno = 28, error message = 'No space left on device', buf = 0x287b17a08, total write size = 8192, bytes this sub-write = 8192, bytes actually written = 18446744073709551615, offset = 0)
Traceback (most recent call last):
  File "h5py/_objects.pyx", line 201, in h5py._objects.ObjectID.__dealloc__
OSError: [Errno 28] Driver write request failed (file write failed: time = Tue Jul 11 19:55:37 2023
[... cut ...]
OSError: [Errno 28] Driver write request failed (file write failed: time = Tue Jul 11 19:55:37 2023
, filename = '/Volumes/test/Kevin/Lungww/MV3ww/GFPww/2023-07-11/Cell_004/CH00_000000.h5', file descriptor = 221, errno = 28, error message = 'No space left on device', buf = 0x287b19c08, total write size = 8192, bytes this sub-write = 8192, bytes actually written = 18446744073709551615, offset = 0)
```

Add apparently, for N5, it does fail gracefully already...
[image]

A proper solution will need to handle each image writing class. I can continue working on this.

zacsimile, 2023-07-12:
My hope is will be as easy as a call to `close()` when we near the limit of the hard drive. Just need to make sure we call `_check_shape()` and close out the data source prior to making a write attempt that exceeds the HD limits.

zacsimile, 2023-07-12:
Actually it might be as easy as catching the `OSError` and calling `close()` in the event it is caught.

closed 2023-07-14 by zacsimile