TheDeanLab/navigate #485: Strange GUI error in the course of running the program, not sure what caused it, happened during ChangeResolution

zacsimile, 2023-06-15:
```
Traceback (most recent call last):
  File "C:\Users\DeanLab\.conda\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\DeanLab\.conda\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\deanlab\desktop\github\aslm\src\aslm\controller\controller.py", line 961, in update_event
    self.update_stage_controller_silent(value)
  File "c:\users\deanlab\desktop\github\aslm\src\aslm\controller\controller.py", line 926, in update_stage_controller_silent
    self.stage_controller.set_position_silent(stage_gui_dict)
  File "c:\users\deanlab\desktop\github\aslm\src\aslm\controller\sub_controllers\stage_controller.py", line 305, in set_position_silent
    self.unbind_position_callbacks()
  File "c:\users\deanlab\desktop\github\aslm\src\aslm\controller\sub_controllers\stage_controller.py", line 250, in unbind_position_callbacks
    self.widget_vals[axis].trace_remove("write", cbname)
  File "C:\Users\DeanLab\.conda\envs\ASLM\lib\tkinter\__init__.py", line 430, in trace_remove
    self._tk.deletecommand(cbname)
_tkinter.TclError: can't delete Tcl command
```

AdvancedImagingUTSW, 2023-06-15:
Log file? But does seem tk specific…

zacsimile, 2023-06-15:
OK this happens when we change resolution on the Mesoscale side and we are not in continuous mode. If we are in continuous mode, there is no issue.

zacsimile, 2023-06-15:
Annoyingly, this seems to kill a thread.

zacsimile, 2023-06-15:
[model_debug.log](https://github.com/TheDeanLab/ASLM/files/11751802/model_debug.log)
[view_controller_debug.log](https://github.com/TheDeanLab/ASLM/files/11751803/view_controller_debug.log)

commit abf0b726 (2023-06-15): fix #485
closed 2023-06-15 by zacsimile