TheDeanLab/navigate #488: Redundant Scan Axis

AdvancedImagingUTSW, 2023-06-16:
I have added the software to control one of our microscopes that we regularly use (as opposed to the developmental systems, this one is production). This microscope does not have a focusing or a rotational stage, but instead has:

- A mechanical stage (Sutter MP285) which positions the entire specimen in X, Y, and Z, for tiling.
- A piezo stage (GalvoNIStage) which is mounted to the MP285, and scans the specimen in Z for acquisition of a Z-stack.

What would you recommend that I do to operate only the piezo for the z-stack, but use the MP285 for the tiling? @zacsimile  thought I should call it into the software as a focusing stage, which would work with everything we have now. I thought I could perhaps define a new axis, called S (for sample scanning), and then modify the z-stack feature to use the S-axis if a stage is defined that has it, otherwise to use what is called Z. Zach preferred not to mess with the z-stack feature.

If we call it in as a focusing stage, I am concerned that the tiling wizard will not work since it needs to know what the size of the z-stack is in order to calculate the grid that the MP285 XYZ stage will use.  If we call in the piezo as a z-stage, then the MP285 has to be gridded out in focusing. Both seem awkward to me.  Tiling is a very important feature on these microscopes, and CT-ASLM-V2 will also have a similar problem.

AdvancedImagingUTSW, 2023-06-16:
I will need to make sure if the remote windows desktop is functioning, but it is this computer (typos included in the microscope name are correct): 
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

CT-ASLM V1 | ND11.106 | 129.112.108.21 | CTASLMv1\ALSM_MASTER |  
-- | -- | -- | -- | --

</html>

zacsimile, 2023-06-16:
If I understand correctly, you want to be able to:

* Open the tiling wizard and define a list of positions `x`, `y`, `s` (where `s` is the new axis that moves in the z-direction)
* Run through these positions and at each one take a z-stack using `z` (the z-piezo)

I think we could use map the sample-scanning stage (what you call `s`) to `f` and the z-piezo to `z`. My thinking here is that the multiposition table simply moves through all positions and `f` appears to the program the same way an `s` axis would. The only thing we have to watch out for is the couple of special things we do with `f` in the z-stack, but this could easily be solved by setting `f` start/stop to 0.

We could add the ability to grid in `f` space to the tiling wizard. In fact it makes sense to allow for each axis to be gridded and then you could optionally check "Fix this axis" if you did not want to grid said axis.

By using `f` instead of `s`, we avoid needing to add another stage control to the GUI, etc. (all the other things that come with a stage). If calling it `f` instead of `s` feels weird, we could consider adding nicknames for stage axes to `configuration.yaml` and pulling these into the GUI.

zacsimile, 2023-06-16:
As discussed in our software meeting, both `z` and `f` canonically move along the same axis, with `z` moving the sample toward/away from the objective and `f` moving the objective toward/away from the sample. If we simply map the sample stage `s` to `z` and the piezo z-movement to `f`, this is consistent. We then need to modify `ZStackAcquisition` slightly to allow for `f` to ramp even when `z` does not. Currently, `z` can ramp while `f` remains fixed or `f` and `z` can ramp together, but `f` cannot ramp independently. Once it can, I believe this will allow for exactly the tiling approach desired here with no further modification of the code.