TheDeanLab/navigate #494: Field of View Not Calculated Properly on Alternative System

AdvancedImagingUTSW, 2023-06-25:
In the `camera_setting_controller` we have some hard-coded calculations that are disrupting our field of view calculation.

```
 # magnification == 'N/A' is a proxy for resolution == 'high'
        if (
            self.parent_controller.configuration["experiment"]["MicroscopeState"][
                "zoom"
            ]
            == "N/A"
        ):
            # 54-12-8 - EFLobj = 12.19 mm / RI
            tube_lens_focal_length = 300
            extended_focal_length = 12.19
            if self.solvent == "BABB":
                refractive_index = 1.56
            elif self.solvent == "Water":
                refractive_index = 1.333
...
```

My first approach was to name our zoom something other than `N/A` in the configuration, experiment, and waveform_constants yaml files. However, when I did this, I get the following error, and I can't seem to figure out where the `N/A` is being pulled from.

```
Exception inside ObjectInSubprocess: Traceback (most recent call last):
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\concurrency\concurrency_tools.py", line 521, in _child_loop
    result = getattr(obj, method_name)(*args, **kwargs)
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\model.py", line 523, in run_command
    waveform_dict = self.active_microscope.calculate_all_waveform()
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\microscope.py", line 383, in calculate_all_waveform
    remote_focus_waveform = self.remote_focus_device.adjust(readout_time)
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\devices\remote_focus\remote_focus_ni.py", line 166, in adjust
    waveform_dict = super().adjust(readout_time)
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\devices\remote_focus\remote_focus_base.py", line 185, in adjust
    temp = waveform_constants["remote_focus_constants"][imaging_mode][zoom][
  File "<string>", line 2, in __getitem__
  File "C:\Users\ALSM_Master\.conda\envs\ASLM\lib\multiprocessing\managers.py", line 825, in _callmethod
    raise convert_to_error(kind, result)
KeyError: 'N/A'

model thread exception happened! 'N/A' Traceback (most recent call last):
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\controller\thread_pool.py", line 175, in func
    target(*args, **kwargs)
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\controller\controller.py", line 526, in <lambda>
    "model", lambda: self.model.run_command("update_setting", "resolution")
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\concurrency\concurrency_tools.py", line 412, in attr
    return _get_response(self, True)
  File "c:\users\alsm_master\documents\github\aslm\src\aslm\model\concurrency\concurrency_tools.py", line 433, in _get_response
    raise resp
Exception: 'N/A'

model thread ended because of exception!: not all arguments converted during string formatting
```

Otherwise, @zacsimile, rather than doing a np.rot90 on the camera, I simply switched what the X and Y axes were in the configuration file and now the image shows up in the correct direction once loaded in BigStitcher, but our overlap is insane since it thinks the field of view is ~340 microns, not ~900.

AdvancedImagingUTSW, 2023-06-25:
Ugh. It was clearly coming from the experiment.yaml file, but I got tricked. I think I updated the experiment file when the program was still open. Thus, when we closed the program, the current microscope state was saved, and experiment file was overwritten. It is working now with the pixel information that I gave it. But we should fix the camera_setting_controller nonetheless. So, this issue remains...

zacsimile, 2023-07-28:
Fixed in #552. I think the solution is to use measured pixel sizes to calculate the FOV. This will involve extending the `pixel_size` properties in `configuration.yaml` from 

```
pixel_size:
        0.63x: 9.7
        1x: 6.38
        2x: 3.14
        3x: 2.12
        4x: 1.609
        5x: 1.255
        6x: 1.044
```

to something like

```
pixel_size:
    BABB:
        0.63x: 9.7
        1x: 6.38
        2x: 3.14
        3x: 2.12
        4x: 1.609
        5x: 1.255
        6x: 1.044
    water:
        0.63x: 10.31
        1x: 6.5
        2x: 3.25
        3x: 2.16
        4x: 1.63
        5x: 1.3
        6x: 1.08
```

zacsimile, 2023-11-04:
Re-upping this as it's important for tiling properly. Do we prefer

```
pixel_size:
  BABB:
    0.63x: 9.7
    1x: 6.38
  water:
    0.63x: 10.31
    1x: 6.5
```

or

```
pixel_size:
  0.63x:
    BABB: 9.7
    water: 10.31
  1x:
     BABB: 6.38
     water: 6.5
```

or something else?