TheDeanLab/navigate #498: Fix Hot Keys

AdvancedImagingUTSW, 2023-06-30:
Control + A should not be autofocus, we will have to designate as something else.

When the save popup dialog is up, keyboard hotkeys should be disabled.

Anytime we are dealing with a text box, no w, a, s, d...

AdvancedImagingUTSW, 2023-07-01:
Few notes here...  

We can create a branch directly from the issue (on the right side of the issues page), which makes it very easy to follow progress. For example, when you create a pull request from a branch that is linked to an issue, it automatically closes the issue. Pretty convenient.  I did this already, and the branch is called `498-fix-hot-keys`

AdvancedImagingUTSW, 2023-07-01:
After creating the branch, you can check out to it locally on your computer. I am lazy, and I use the GitHub Desktop App. It simplifies misc annoying errors you experience along the way. 

For ASLM, the hotkeys are largely defined in one of two places:
`menu_controller.py`
`keystroke_controller.py`

I just pushed a slightly better numpydoc explanation of how the menu_controller specifies hotkeys.  I will also modify the autofocus to be control + shift + a to immediately run the autofocus routine, and control + shift + alt + a to open the popup window for the autofocus routine.

mehr0096, 2023-07-01:
Thank you for the information Kevin. I will go through these files to disable the w a s d when a textbox shows up and disable all hotkeys when the popup box shows up. I will work in the 498-fix-hot-keys branch.

AdvancedImagingUTSW, 2023-07-01:
No worries, it is just nap time, and I am killing time, so figured I would give you a walkthrough of the code to some extent...

It continues...

To see how we can make sure that hotkeys are not called if we have certain types of popup windows open, I called the following within the `stage_key_press` function in the `stage_controller`.

```
print("GUI FOCUS!!!", self.parent_controller.view.focus_get())
```

After pressing a stage movement hotkey (WASD), this lets me know that the save dialog is somehow called .!popup, and the multi position acquisition popup is called .!popup2. 

Just to make sure that wasn't reflective of the order I opened them, I tried it in reverse, and sadly, it now called the multi position acquisition popup .~popup... So, it seems to name it in the order that it is called. I assume we want the hotkeys to function with the multi position popup, but not the save popup.

closed 2023-07-12 by zacsimile