TheDeanLab/navigate #520: Add flipx/flipy/flipz flags to `configuration.yaml` for stages

zacsimile, 2023-07-14:
Assume by default that increasing X, Y, or Z moves the sample closer to the objectives (further into the field of view). Use this information when saving positions of tiles during acquisition so that the tiled image comes out in the right order. Currently, this is done for `VolumeSearch` and can be changed on a per-scope basis using the plugin architecture for features. We should generalize this behavior.

AdvancedImagingUTSW, 2023-07-31:
This will have some serious ramifications for features and microscope operation.

My interpretation is that when we do flipX = True, then essentially, when we normally tell the stage to move + 500 microns, it would move -500 microns. 

Alternatively, we could consider multiplying our absolute stage position by negative 1.  

What exactly did you have in mind @zacsimile ?

zacsimile, 2023-07-31:
Your interpretation is correct. Right now I implement it with a sign flip in `VolumeSearch`, which is effectively the same as saying move `-500` instead of `+500`.

closed 2023-08-07 by AdvancedImagingUTSW