TheDeanLab/navigate #521: Add flipx/flipy flags to `configuration.yaml` for camera, act accordingly

zacsimile, 2023-07-14:
The camera may be flipped w.r.t. the normal stage orientations (see #520). It may be the case (as in the ctASLMv2) that we need to not only track positions but flip the pixel orientation (the ctASLMv2 camera is upside-down and needs to be saved to disk in reverse order, `img[:,::-1]`).

AdvancedImagingUTSW, 2023-07-31:
Discussing this with @annie-xd-wang 

I assume that what we show the user in the GUI should also have been flipped in x or y, depending upon what the flags state. Is that correct @zacsimile?

No rotation operations?

zacsimile, 2023-07-31:
I think we can optionally turn this off or on in the GUI with a button, the same way we do with Flip XY at the moment. I haven't run into a rotation requirement yet, but we certainly could implement one in case a camera is installed at a different angle.

closed 2023-08-09 by AdvancedImagingUTSW