TheDeanLab/navigate #525: SNR calculation is very slow, makes continuous mode appear delayed

zacsimile, 2023-07-17:
As the title states, the SNR processing makes the data thread take a much longer time to get through displaying all of the images. This leads to non-real-time feedback.

zacsimile, 2023-07-26:
@annie-xd-wang more or less circumvented this problem in #533. Converting everything to float on initial loading may help. The `sqrt()` is likely slowing us down a lot. It's kind of an expensive calculation. Will have a think.

AdvancedImagingUTSW, 2023-07-26:
Quickly tried some other ways to do the square root operation, but numpy is pretty damn good. 22x faster than the built in python method for a 2.2k x 2.2k 16-bit image.

```
Python Version - Duration: 0.044397497177124025
Numpy Version - Duration: 0.002041006088256836
```

closed 2026-02-28 by AdvancedImagingUTSW