TheDeanLab/navigate #534: ctASLMv2_Laser_Issue

JinlongL, 2023-07-19:
1. When acquiring Z stack with 3 channels, 647 channel laser wont turn on. But if just have 647 channel selected, the laser will work normally.  
2. When imaging with more than 1 channel, the laser power of the first channel will be normal, but the power of the second channel will stay low. To check if the laser power of the second channel is the same as the first channel, we changed the laser power of the first channel significantly, the laser power on the second laser still stay the same.
[image]

zacsimile, 2023-07-20:
This appears to be related to the following error.

```
2023-07-20 17:22:58,736 - model - ERROR - laser_ni: The specified resource is reserved. The operation could not be completed as specified.
Task Name: _unnamedTask<6>

Status Code: -50103
Traceback (most recent call last):
  File "c:\users\spectral\documents\github\aslm\src\aslm\model\devices\lasers\laser_ni.py", line 103, in set_power
    self.laser_ao_task.write(scaled_laser_voltage, auto_start=True)
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\nidaqmx\task.py", line 1233, in write
    return _write_analog_f_64(
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\nidaqmx\_task_modules\write_functions.py", line 32, in _write_analog_f_64
    check_for_error(error_code)
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: The specified resource is reserved. The operation could not be completed as specified.
Task Name: _unnamedTask<6>
```

I cannot replicate this error on other microscopes.

zacsimile, 2023-07-20:
This problem is resolved if we disable `create_analog_output_tasks()`. I believe this suggests the PCIe-6738 can handle fewer analog output tasks than the PXI-6259.

zacsimile, 2023-07-20:
Confirmed. The PCIe-6738 can only create one software-timed analog task for every four channels (see https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019KWYSA2&l=en-US). The PXI-6259 can create one software-timed analog task per channel.

Rather than fixing this in the software, which would basically involve pushing everything to a single AO write as we do for the galvos/remote focus units, I propose we attach lasers to analog output ports outside of the banks used by the galvo/remote focus units. For example, on the v2, we use `ao`s `0`, `2`, `6` for the remote focus, galvo and galvo stage. We use `1`, `3`, `4` for the lasers. As seen in the attached pinout, `ao`s `1` and `3` are on the same bank as `0` and `2`, and `ao4` is on the same bank as `ao6`. If we move the lasers to `ao`s `8`, `9`, `10`, they will compete with none of the other `ao` ports. Since only one task will be created created on the `8`,`9`,`10` `ao` bank at a time (only one laser is on at a time), we can get away with this. If we wanted to turn lasers on simultaneously, we could distribute the lasers across independent banks (e.g. `ao8`, `ao14`, `ao19`).

zacsimile, 2023-07-20:
Confirmed. The 8/9/10 rewiring solves it on the v2. We will have to do a similar rewiring on the v1 and the new mesoSPIM. https://github.com/TheDeanLab/ASLM/assets/1263313/6605abd1-d242-4469-8ee7-c88a05d29b56

AdvancedImagingUTSW, 2023-07-21:
Nice find @zacsimile. I learned something, that one was very non-intuitive.

AdvancedImagingUTSW, 2023-07-21:
I updated the documentation to basically paraphrase what you state above, and pushed it to develop.

zacsimile, 2023-07-26:
This is addressed by the new documentation and I have made the changes on the v2 and the mesoSPIM. The v1 still needs to be rewired.

closed 2023-07-26 by zacsimile

AdvancedImagingUTSW, 2023-07-26:
Perhaps @JinlongL could rewire CT-ASLMv1?