TheDeanLab/navigate #543: Acquiring with N5 hangs program indefinitely

zacsimile, 2023-07-24:
This is on the new mesoSPIM with the Tiger stage. No error is thrown, but the acquisition seems to hang indefinitely. Could just be a slow save.

AdvancedImagingUTSW, 2023-08-06:
Perhaps we could parallelize the following lines (beginning at line 137) in the `bdv_data_source`. It could provide us with a 2x or 3x gain, depending on how much time it takes to spin up the down different threads/pool objects.

```
        for i in range(self.subdivisions.shape[0]):
            dx, dy, dz = self.resolutions[i, ...]
            if z % dz == 0:
                dataset_name = ds_name.replace("???", str(i))
                # print(z, dz, dataset_name, self.image[dataset_name].shape,
                #       data[::dx, ::dy].shape)
                zs = np.minimum(
                    z // dz, self.shapes[i, 0] - 1
                )  # TODO: Is this necessary?
                self.image[dataset_name][zs, ...] = data[::dx, ::dy]
                if (i == 0) and len(kw) > 0:
                    self._views.append(kw)
```

What do you think @annie-xd-wang

AdvancedImagingUTSW, 2023-08-07:
N5/HDF5/Zarr all do this down-sampling in what is called a 'pyramidal file format'. So, lots of people are doing this, and there are probably much more optimized methods for doing this.

For example, Zarr. 
https://github.com/zarr-developers/zarr-python

This paper also does it:
https://www.biorxiv.org/content/10.1101/2023.06.08.544277v3.full.pdf

In the N5 file formats, each downsampled chunk of data is saved in its own directory. This seems very likely to be ideal for thread-based saving protocols since that chunk is independent of every other chunk.

AdvancedImagingUTSW, 2023-08-07:
```
To achieve robust and reliable acquisition at the speed required by the ExA-SPIM, we used two data streaming
strategies. The first used the open-source eGrabber frame grabber Python API10 and open-source ImarisWriter library (Beati et al. 2020). This option enabled high-speed data streaming with online lossless compression (LZ4 with bit shuffling) and real-time writing of a multi-resolution pyramid, both of which help streamline downstream data storage and handling. The second strategy used Acquire (https://github.com/acquire-project), led by the Chan Zuckerberg Initiative Imaging team in collaboration with the Allen Institute for Neural Dynamics (Clack et al. 2023). Acquire is a new state-of-the-art microscope acquisition project which enables the ExA-SPIM system to stream data at the required rate, directly to OME-Zarr, with either ZStandard or LZ4 compression, and a variable chunk size.
```

zacsimile, 2023-08-20:
We certainly could (and probably should) parallelize the multi-resolution writes. `zarr` supports this and `h5py` seems to have some support via MPI (though I'm worried this will be a headache, and potentially unnecessary if we have parallel write in N5 or Zarr already). 

I was able to speed the N5 write up about 50x in #594 by adjusting the chunk size to match the size of the frame we are writing to disk. Hint from https://forum.image.sc/t/slow-writing-to-persistent-zarr-array/59556. Proof in `cProfile`:

```
tuple(chunks)
Conditions are multiposition: False per_stack: True z_stack: True stop_early: False
x: 1024 y: 2048 z: 3 c: 3 t: 2 positions: 1 per_stack: True
         9448971 function calls (9437372 primitive calls) in 42.934 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
    49072   13.872    0.000   13.872    0.000 {built-in method posix.replace}
    49591   11.197    0.000   11.197    0.000 {built-in method io.open}
    16714    3.503    0.000    3.503    0.000 {built-in method posix.mkdir}
    49591    3.169    0.000    3.169    0.000 {method '__exit__' of '_io._IOBase' objects}
   262670    2.659    0.000    2.659    0.000 {built-in method posix.stat}
    48960    0.756    0.000    2.015    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/zarr/n5.py:872(encode)
    49072    0.574    0.000   35.446    0.001 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/zarr/storage.py:1089(__setitem__)
    48960    0.340    0.000    2.191    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/zarr/core.py:2150(_process_for_setitem)
   147823    0.312    0.000    0.474    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/posixpath.py:71(join)
   927050    0.242    0.000    0.505    0.000 {built-in method builtins.isinstance}
```

vs

```
(shape[0], shape[1], 1)
Conditions are multiposition: False per_stack: True z_stack: True stop_early: False
x: 1024 y: 2048 z: 3 c: 3 t: 2 positions: 1 per_stack: True
         166083 function calls (155012 primitive calls) in 0.859 seconds

   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
      703    0.444    0.001    0.444    0.001 {built-in method io.open}
      184    0.086    0.000    0.086    0.000 {built-in method posix.replace}
       72    0.056    0.001    0.056    0.001 {method 'byteswap' of 'numpy.ndarray' objects}
       72    0.039    0.001    0.098    0.001 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/zarr/n5.py:872(encode)
      703    0.024    0.000    0.024    0.000 {method '__exit__' of '_io._IOBase' objects}
     1598    0.021    0.000    0.021    0.000 {built-in method posix.stat}
       82    0.020    0.000    0.020    0.000 {built-in method posix.mkdir}
       72    0.018    0.000    0.031    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/zarr/core.py:2150(_process_for_setitem)
       72    0.010    0.000    0.010    0.000 {method 'fill' of 'numpy.ndarray' objects}
26631/17031    0.009    0.000    0.017    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/json/encoder.py:333(_iterencode_dict)
      184    0.008    0.000    0.008    0.000 {method 'write' of '_io.BufferedWriter' objects}
```

zacsimile, 2023-08-21:
It seems that even with the speedup, it is a problem on the mesoSPIM v2. It appears to be linked with the stage delay. Things work fast enough on the multiscale.

[model_debug.log](https://github.com/TheDeanLab/ASLM/files/12402025/model_debug.log)

zacsimile, 2023-08-21:
Ok, the N5 is now fast, but the H5 is still slow. Perhaps we should chunk the H5 similarly to the N5.

zacsimile, 2023-08-21:
OK, nope. It just hates the H5 on this one machine. Chunking doesn't help. N5 runs fine now, though.

closed 2023-08-21 by zacsimile