TheDeanLab/navigate #548: Predictable BigTiff vs Tiff Error

zacsimile, 2023-07-27:
```
Exception in thread z-stack Data:
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\tifffile\tifffile.py", line 2873, in _write_remaining_pages
    ifd.write(pack(offsetformat, fhpos + pos))
struct.error: argument out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\spectral\documents\github\aslm\src\aslm\model\model.py", line 738, in run_data_process
    data_func(frame_ids)
  File "c:\users\spectral\documents\github\aslm\src\aslm\model\features\image_writer.py", line 230, in save_image
    self.data_source.write(
  File "c:\users\spectral\documents\github\aslm\src\aslm\model\data_sources\tiff_data_source.py", line 164, in write
    self.close(True)
  File "c:\users\spectral\documents\github\aslm\src\aslm\model\data_sources\tiff_data_source.py", line 239, in close
    self.image[ch].close()
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\tifffile\tifffile.py", line 3030, in close
    self._write_remaining_pages()
  File "C:\Users\Spectral\.conda\envs\ASLM\lib\site-packages\tifffile\tifffile.py", line 2881, in _write_remaining_pages
    raise ValueError('data too large for non-BigTIFF file')
ValueError: data too large for non-BigTIFF file

```

zacsimile, 2023-07-28:
A solution would be to use `tiff_data_source.size()` to calculate the expected size and set `tiff_data_source.is_bigtiff` to `True` if you are saving more than 4 GB of data.

AdvancedImagingUTSW, 2023-07-28:
Exactly the approach I had in mind!

closed 2023-08-04 by JinlongL