TheDeanLab/navigate #569: New_mesoSPIM_Multi-position_acquisition_stopped_in_the_middle

JinlongL, 2023-08-03:
While trying to perform z stack on multi position, software stopped while only go through half of the positions. In the prompt shows: NIDAQMX error. Log attached.   Then tried to run continues scan, anaconda and ASLM software froze. 
[model_debug.log](https://github.com/TheDeanLab/ASLM/files/12255683/model_debug.log)
[view_controller_debug.log](https://github.com/TheDeanLab/ASLM/files/12255684/view_controller_debug.log)

JinlongL, 2023-08-04:
```
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
Exception in thread z-stack signal:
Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\model.py", line 922, in run_acquisition
    self.snap_image()
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\model.py", line 888, in snap_image
    self.active_microscope.daq.run_acquisition()
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\devices\daq\daq_ni.py", line 427, in run_acquisition
    if self.camera_trigger_task.is_task_done():
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\site-packages\nidaqmx\task.py", line 520, in is_task_done
    check_for_error(error_code)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Task specified is invalid or does not exist.
Status Code: -200088
```

AdvancedImagingUTSW, 2023-08-04:
I am wondering if we are getting to a part of the multi-position acquisition where the move is large, and is taking long enough for the camera to time out. Could you also upload the experiment file that was saved with the image data?

JinlongL, 2023-08-04:
Here is the file. The same error happened a few more times when I tried to image the remaining locations. What weird was I was able to run the acquisition over the same location sequences again and had better luck the second time.  [experiment - Copy.zip](https://github.com/TheDeanLab/ASLM/files/12263850/experiment.-.Copy.zip)

AdvancedImagingUTSW, 2023-08-04:
It looks like the z stage is trying to move about 4 mm...

```2023-08-04 12:25:16,283 - model - DEBUG - common_features: *** Zstack move stage: (z: 3867.20000000014), (f: 1032.6)
2023-08-04 12:25:26,757 - model - DEBUG - common_features: *** Zstack move stage: (z: -932.5), (f: 1032.6)
```

JinlongL, 2023-08-04:
Different run
```
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
Exception in thread z-stack signal:
Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\model.py", line 922, in run_acquisition
    self.snap_image()
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\model.py", line 888, in snap_image
    self.active_microscope.daq.run_acquisition()
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\devices\daq\daq_ni.py", line 427, in run_acquisition
    if self.camera_trigger_task.is_task_done():
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\site-packages\nidaqmx\task.py", line 520, in is_task_done
    check_for_error(error_code)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Task specified is invalid or does not exist.
```
[view_controller_debug.log](https://github.com/TheDeanLab/ASLM/files/12264001/view_controller_debug.log)
[model_debug.log](https://github.com/TheDeanLab/ASLM/files/12264002/model_debug.log)
[experiment - Copy.zip](https://github.com/TheDeanLab/ASLM/files/12264007/experiment.-.Copy.zip)

AdvancedImagingUTSW, 2023-08-05:
According to @JinlongL, we have not seen the error since we implemented this fix.

closed 2023-08-05 by AdvancedImagingUTSW