TheDeanLab/navigate #592: (ctASLMv1)Camera starts acquiring data before stage moves to starting position

JinlongL, 2023-08-18:
While doing Z stack acquisition, user moves stage to the starting position, mark it as start pos, then move to the end position, and mark it as end pos. Without bringing the z piezo back to the starting position before imaging, end position will be the first frame. In single channel image, we can bring the z piezo back to the starting position before the acquisition, so the start position will be the first frame. However, when doing multi-channel imaging, bring the z piezo back to the start position before starting the whole acquisition will only have the start position in the first frame in the first channel, other channels will still have the end position in the first frame. The frame number or the slice number will be the same for all channels, resulting the n frame of the first channel be the n+1 frame for other channels. 
[image]

zacsimile, 2023-08-18:
Which system?

JinlongL, 2023-08-18:
The V1. Not a priority. But just to document it.

EXTERNAL MAIL

CAUTION: This email originated from outside UTSW. Please be cautious of links or attachments, and validate the sender's email address before replying.

UT Southwestern

Medical Center

The future of medicine, today.

zacsimile, 2023-08-19:
I believe this has to do with the response time of the z-piezo on this system. It should also be a problem on the v2. We could consider adding a settling time parameter to `GalvoNIStage`. This will slow down acquisition. Alternatively, we could investigate if the voltage-controlled piezos on those systems output an on target signal and look for that.

For multi-channel acquisition, you could work around this problem by bringing the piezo to the starting z-position and acquiring `per_z` instead of `per_stack`.

AdvancedImagingUTSW, 2023-10-27:
@zacsimile - I tried to implement a fix for this, but strangely it doesn't seem to work. Perhaps it will be an obvious thing to you.

Essentially, when we execute the command `move_absolute` in the `GalvoNIStage`, we now have a distance threshold and a settling duration implemented. These values are pulled with getters from the configuration file. 

However, even when implementing what I consider a very long settling duration (50 milliseconds), the saved image still shows the wrong position of the tissue in the first frame. I have confirmed that the sleep call is being triggered as expected. But it is almost as if the camera is grabbing an image before the `move_absolute` is being called.

I have this pulled on CT-ASLM-V2 and V1.

AdvancedImagingUTSW, 2023-10-31:
So the stage delay doesn't fix anything. Good thing to have implemented nonetheless, given that you can see the settling time on the oscilloscope.

However, one hint that I am trying to look into is that the starting position of the stage changes the behavior. Everything below is for a 200 micron z-stack. 

Z-stack starting position: 200 microns. Voltage resting at 9V, jumps up to 10V, captures an image, swings down to 1V, then continues a z-stack as normal.

Z-stack starting position: 150 microns. Voltage resting at 9V, jumps to ~7.8V, captures an image, swings down to 1V, ...

Z-stack starting position: 100 microns. Voltage resting at 9V, jumps to 5V, captures in image, swings down to 1V, ...

Z-stack starting position: 50 microns. Voltage resting at 9V, jumps to 2.5V, captures in image, swings down to 1V....

Z-stack starting position: 0 microns. Works as intended. 

So, somewhere in the z-stack function we are going to the current z-position, not the starting z-position.

AdvancedImagingUTSW, 2023-10-31:
With some help, managed to track down where this is coming from. 

model.prepare_acquisition() ...
model.active_microscope.prepare_acquisition() ...
model.active_microscope.calculate_all_waveform() ...

Key if statement on line 426 (if type(stage) == galvoNIStage)
model.active_microscope.stage.calculate_waveform

Essentially, the prepare_acquisition call results in a false start. All of the waveforms, including the camera/galvo/NIstage/etc., are delivered off the DAQ and an image is captured. 

Whil;e trying to kludge my way around it, I tried the following in `galvoNIStage` for the method `calculate_waveform`: 
```
        if self.configuration["experiment"]["MicroscopeState"]["image_mode"] == "z-stack":
            # If we are in the z-stack mode, we should move the stage to the start position
            volts = eval(
                self.volts_per_micron,
                {"x": self.configuration["experiment"]["MicroscopeState"]["start_position"]},
            )
            # start_position
        else:
            # If we are in the live mode, we want to keep the stage where we are.
            volts = eval(
                self.volts_per_micron,
                {"x": self.configuration["experiment"]["StageParameters"][self.axes[0]]},
            )
```
This at least gets us to the right position for starting the z-stack, but the indexing error still shows up as an extra frame when we acquire data in a per_z mode.

AdvancedImagingUTSW, 2023-10-31:
One last note.  

In `model.snap_image()`, we could also circumvent the problem by running the signal_container twice. However, this failed if we did a multi channel acquisition.

```        
if hasattr(self, "signal_container"):
            self.signal_container.run()
            # self.run_uno += 1
            #
            # if self.imaging_mode == "z-stack":
            #     if self.run_uno == 1:
            #         self.signal_container.run()
```

AdvancedImagingUTSW, 2023-10-31:
Per stack mode.starting position between stacks is off since an additional frame is acquired between stacks.

 [image]

AdvancedImagingUTSW, 2023-11-06:
It sounds like Annie will be back on the 27th, so unless this is more pressing, we can wait for her to handle it then.

AdvancedImagingUTSW, 2023-11-28:
@annie-xd-wang provided a couple of tests, which I ran on CT-ASLM-V2. They are located [here](https://github.com/TheDeanLab/archicellago/blob/main/src/archicellago/hardware/2023_11_28_Daq_NIStage_task_write_test.ipynb).

In scenario 1, the DAQ will first go to the position listed in the experiment.yaml file, then go to each position in the list of positions that we provide. This does occur, which means that the daq.task.write() queues the waveform instead of overrides it.

In scenario 2, the DAQ only goes through the positions in the list of positions that we provide. This is also true.

AdvancedImagingUTSW, 2023-11-29:
We approached this issue from two separate directions. 

The first was that the DAQ was queuing waveforms, not overwriting them. This is the PR that @annie-xd-wang just made, and which appears functional now.

The second was if the stage was still moving when we acquired the image, which could result in the image appearing blurry. This I approached by implementing a threshold for a movement that would trigger a wait duration, and this function is being called when I expect it now. However, the vibration in the specimen still exists no matter how long I set the settle duration to, and I think I know why... 

In the `GalvoNIStage`, the trigger source is still active. Thus, the stage will only move when the external trigger has told it to, and then we wait after that point. For the GalvoNIStage, it seems as if the analog output should be delivered instantly and not triggered. It also presumably doesn't need to be a `dc_value` but rather just a single float value.

```        
self.daq.analog_outputs[self.axes_channels[0]] = {
            "sample_rate": self.sample_rate,
            "samples": self.samples,
            "trigger_source": self.trigger_source,
            "waveform": self.waveform_dict,
        }
        # update analog waveform
        self.daq.update_analog_task(self.axes_channels[0].split("/")[0])

        # Stage Settle Duration in Milliseconds
        if (
            wait_until_done
            and (self.distance_threshold is not None)
            and (delta_position >= self.distance_threshold)
        ):
            time.sleep(self.stage_settle_duration)
```

AdvancedImagingUTSW, 2023-11-29:
Naively, it seems like we should be able to create a Task directly within the GalvoNIStage class that is of type `stream_writers.AnalogMultiChannelWriter`, and then write a single voltage out using `write_one_sample`

https://nidaqmx-python.readthedocs.io/en/latest/stream_writers.html

This would eliminate all of this code:

```
  # channel includes 'is_selected', 'laser', 'filter', 'camera_exposure'...
            channel = microscope_state["channels"][channel_key]

            # Only proceed if it is enabled in the GUI
            if channel["is_selected"] is True:

                # Get the Waveform Parameters
                # Assumes Remote Focus Delay < Camera Delay.  Should Assert.
                try:
                    self.sweep_time = self.sweep_times[channel_key]
                except TypeError:
                    # In the event we have not called calculate_waveform in advance...
                    # Assumes Remote Focus Delay < Camera Delay.  Should Assert.
                    exposure_time = channel["camera_exposure_time"] / 1000
                    self.sweep_time = exposure_time + exposure_time * (
                        (self.camera_delay_percent + self.remote_focus_ramp_falling)
                        / 100
                    )

                    duty_cycle_wait_duration = (
                        float(
                            self.configuration["waveform_constants"]
                            .get("other_constants", {})
                            .get("remote_focus_settle_duration", 0)
                        )
                        / 1000
                    )

                    self.sweep_time += duty_cycle_wait_duration

                self.samples = int(self.sample_rate * self.sweep_time)

                # Calculate the Waveforms
                self.waveform_dict[channel_key] = dc_value(
                    sample_rate=self.sample_rate,
                    sweep_time=self.sweep_time,
                    amplitude=volts,
                )
```

PR #661 (2023-11-13): Galvo NI stage delay
While trying to fix #592, first tried to implement `wait_until_done` functionality in the GalvoNIStage.  Turns out that #592 is a bit more complicated and has to do with calculating waveforms and an indexing error in the feature container, so I wanted to split this off from that branch and make it a separate PR.  It seems as if this would be valuable for some of the high-speed tiling being performed by @zacsimile where the stage does not appear entirely settled between adjacent z-stacks.

For operation, requires new parameters in the `configuration.yaml` file. Uses getters [... cut]
closed 2023-11-29 by AdvancedImagingUTSW