TheDeanLab/navigate #596: H5 slow to save on the mesoSPIM v2 machine

zacsimile, 2023-08-21:
N5 is speedy on the mesoSPIM v2. H5 is speedy on the multiscale. Not sure what's going on.

AdvancedImagingUTSW, 2023-08-22:
Is N5 also speedy on the multiscale? If so, maybe we just move on without it since Zarr supports n5...

https://github.com/saalfeldlab/n5-utils

https://zarr.readthedocs.io/en/stable/api/n5.html

AdvancedImagingUTSW, 2023-08-22:
Have you dabbled with the core driver?

https://docs.h5py.org/en/stable/high/file.html

https://github.com/h5py/h5py/issues/1827

AdvancedImagingUTSW, 2023-08-22:
Quick test to evaluate the different HDF5 drivers.

<details>
  <summary>Default HDF5 Driver</summary>

Interestingly there is a periodicity to this delay. Every 8th write is slow...

Function 'write' executed in 1.2858s
Function 'write' executed in 0.1316s
Function 'write' executed in 0.1316s
Function 'write' executed in 0.1237s
Function 'write' executed in 0.1277s
Function 'write' executed in 0.1317s
Function 'write' executed in 0.1366s
Function 'write' executed in 0.1437s
Function 'write' executed in 0.5020s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1336s
Function 'write' executed in 0.1536s
Function 'write' executed in 0.1536s
Function 'write' executed in 0.1397s
Function 'write' executed in 0.1496s
Function 'write' executed in 0.1406s
Function 'write' executed in 1.0061s
Function 'write' executed in 0.1706s
Function 'write' executed in 0.1416s
Function 'write' executed in 0.1426s
Function 'write' executed in 0.1207s
Function 'write' executed in 0.1437s
Function 'write' executed in 0.1516s
Function 'write' executed in 0.1416s
Function 'write' executed in 1.0494s
Function 'write' executed in 0.1436s
Function 'write' executed in 0.1246s
Function 'write' executed in 0.1206s
Function 'write' executed in 0.1297s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1576s
Function 'write' executed in 1.4349s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1207s
Function 'write' executed in 0.1366s
Function 'write' executed in 0.1206s
Function 'write' executed in 0.1356s
Function 'write' executed in 0.1535s
Function 'write' executed in 0.1455s
</details>

<details>
  <summary>sec2HDF5 Driver</summary>
Timed out...
</details>

<details>
  <summary>stdio Driver</summary>

Function 'write' executed in 0.8432s
Function 'write' executed in 0.1467s
Function 'write' executed in 0.1426s
Function 'write' executed in 0.1297s
Function 'write' executed in 0.1456s
Function 'write' executed in 0.1476s
Function 'write' executed in 0.1497s
Function 'write' executed in 0.1706s
Function 'write' executed in 1.2012s
Function 'write' executed in 0.1596s
Function 'write' executed in 0.1326s
Function 'write' executed in 0.1277s
Function 'write' executed in 0.1496s
Function 'write' executed in 0.1297s
Function 'write' executed in 0.1506s
Function 'write' executed in 0.1486s
Function 'write' executed in 1.2629s
Function 'write' executed in 0.1427s
Function 'write' executed in 0.1456s
Function 'write' executed in 0.1606s
Function 'write' executed in 0.1346s
Function 'write' executed in 0.1227s
Function 'write' executed in 0.1356s
Function 'write' executed in 0.1426s
Function 'write' executed in 1.0258s
Function 'write' executed in 0.1287s
Function 'write' executed in 0.1277s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1426s
Function 'write' executed in 0.1487s
Function 'write' executed in 0.1376s
Function 'write' executed in 0.1417s
Function 'write' executed in 1.3466s
Function 'write' executed in 0.1505s
Function 'write' executed in 0.1197s
Function 'write' executed in 0.1207s
Function 'write' executed in 0.1207s
Function 'write' executed in 0.1097s
Function 'write' executed in 0.1286s
Function 'write' executed in 0.1376s
</details>

<details>
  <summary>Core Driver</summary>

PermissionError: [Errno 13] Unable to open file (file read failed: time = Tue Aug 22 14:34:02 2023
, filename = 'C:\Users\Dean-Lab\Desktop\Data\HDF5\TEST\M405_AB9553\SytoxGreen-488_antiRFP-CF647\2023-08-22\Cell_005\CH00_000000.h5', file descriptor = 6, errno = 13, error message = 'Permission denied', file->mem = 000001FF22C1EF00, total read size = 65536, bytes this sub-read = 65536, bytes actually read = 18446744073709551615, offset = 0)

During handling of the above exception, another exception occurred:

</details>

Qualitatively, the stdio driver is best. But not much better. Will look into why the 8th image is always ~10x slower...

AdvancedImagingUTSW, 2023-08-22:
Periodicity eliminated by changing the HDF5 chunk size as suggested...

```
                    self.image.create_dataset(
                        dataset_name,
                        chunks=(self.shapes[j, 0], self.shapes[j, 1], 1),
                        # chunks=tuple(self.subdivisions[j, ...][::-1]),
                        shape=self.shapes[j, ...],
                        dtype="uint16",
                    )
```

<details>
  <summary>Timing...</summary>

Function 'write' executed in 1.4391s
Function 'write' executed in 0.3481s
Function 'write' executed in 0.3261s
Function 'write' executed in 0.3251s
Function 'write' executed in 0.3092s
Function 'write' executed in 0.3092s
Function 'write' executed in 0.3052s
Function 'write' executed in 0.3261s
Function 'write' executed in 0.3112s
Function 'write' executed in 0.3231s
Function 'write' executed in 0.3112s
Function 'write' executed in 0.3382s
Function 'write' executed in 0.3291s
Function 'write' executed in 0.3241s
Function 'write' executed in 0.3251s
Function 'write' executed in 0.3351s
Function 'write' executed in 0.3231s
Function 'write' executed in 0.3541s
Function 'write' executed in 0.3451s
Function 'write' executed in 0.3112s
Function 'write' executed in 0.3281s
Function 'write' executed in 0.3142s
Function 'write' executed in 0.3261s
Function 'write' executed in 0.3241s
Function 'write' executed in 0.3271s
Function 'write' executed in 0.3441s
Function 'write' executed in 0.3491s
Function 'write' executed in 0.3322s
Function 'write' executed in 0.3670s
Function 'write' executed in 0.3191s
Function 'write' executed in 0.3251s
Function 'write' executed in 0.3324s
Function 'write' executed in 0.3032s
Function 'write' executed in 0.3072s
Function 'write' executed in 0.3102s
Function 'write' executed in 0.3052s
Function 'write' executed in 0.3311s
Function 'write' executed in 0.3291s
Function 'write' executed in 0.3471s
Closing the file

AdvancedImagingUTSW, 2023-08-22:
Tried to upgrade h5py via pip. No luck....

Installed HDF5 library onto host computer.  https://www.hdfgroup.org/downloads/hdf5/?1692736648
No luck...

<details>
  <summary>Timing...</summary>

Function 'write' executed in 1.1835s
Function 'write' executed in 0.3740s
Function 'write' executed in 0.3400s
Function 'write' executed in 0.3461s
Function 'write' executed in 0.3411s
Function 'write' executed in 0.3531s
Function 'write' executed in 0.3571s
Function 'write' executed in 0.3451s
Function 'write' executed in 0.3441s
Function 'write' executed in 0.3331s
Function 'write' executed in 0.3181s
Function 'write' executed in 0.3191s
Function 'write' executed in 0.3331s
Function 'write' executed in 0.3241s
Function 'write' executed in 0.3162s
Function 'write' executed in 0.3152s
Function 'write' executed in 0.3441s
Function 'write' executed in 0.3381s
Function 'write' executed in 0.3332s
Function 'write' executed in 0.3401s
Function 'write' executed in 0.3201s
Function 'write' executed in 0.3241s
Function 'write' executed in 0.3301s
Function 'write' executed in 0.3361s
Function 'write' executed in 0.3152s
Function 'write' executed in 0.3424s
Function 'write' executed in 0.3411s
Function 'write' executed in 0.3381s
Function 'write' executed in 0.3162s
Function 'write' executed in 0.3172s
Function 'write' executed in 0.3231s
Function 'write' executed in 0.3463s
Function 'write' executed in 0.3223s
Function 'write' executed in 0.3635s
Function 'write' executed in 0.3277s
Function 'write' executed in 0.3481s
Function 'write' executed in 0.3581s
Closing the file

zacsimile, 2023-08-22:
Yeah this is crazy. It's a full order of magnitude slower than on my mac.

```
Function 'write' executed in 0.0349s
Function 'write' executed in 0.0265s
Function 'write' executed in 0.0274s
Function 'write' executed in 0.0262s
Function 'write' executed in 0.0273s
Function 'write' executed in 0.0296s
Function 'write' executed in 0.0300s
Function 'write' executed in 0.0286s
Function 'write' executed in 0.0324s
Function 'write' executed in 0.0338s
Function 'write' executed in 0.0304s
Function 'write' executed in 0.0335s
Function 'write' executed in 0.0276s
Function 'write' executed in 0.0280s
Function 'write' executed in 0.0237s
Function 'write' executed in 0.0233s
Function 'write' executed in 0.0230s
Function 'write' executed in 0.1061s
```

```
   Ordered by: internal time

   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
       72    0.326    0.005    0.338    0.005 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/dataset.py:831(__setitem__)
       18    0.156    0.009    0.592    0.033 /Users/zachmarin/Code/ASLM/src/aslm/model/data_sources/bdv_data_source.py:117(write)
        1    0.082    0.082    0.082    0.082 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/files.py:542(close)
       78    0.009    0.000    0.009    0.000 {built-in method numpy.asarray}
        1    0.004    0.004    0.004    0.004 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/files.py:206(make_fid)
       79    0.002    0.000    0.005    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/group.py:319(__getitem__)
      102    0.001    0.000    0.002    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/dataset.py:614(__init__)
       57    0.001    0.000    0.001    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/group.py:480(__contains__)
       30    0.001    0.000    0.001    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/dataset.py:36(make_new_dset)
      360    0.001    0.000    0.001    0.000 /Users/zachmarin/opt/miniconda3/envs/ASLM/lib/python3.9/site-packages/h5py/_hl/dataset.py:514(dtype)
        1    0.001    0.001    0.001    0.001 {built-in method io.open}
```

AdvancedImagingUTSW, 2023-08-22:
Maybe it’s time to ping Colfax international?

zacsimile, 2023-08-23:
Confirmed that H5 runs normally on the v2 and the upright. It seems to be a problem specific to the mesoSPIM v2 machine.

AdvancedImagingUTSW, 2023-08-23:
After evaluating the read/write times for the hard-drive on a few machines, we identified that we were writing to a hard-drive and not an SSD. The computer had an unformatted SSD, which is now functional. Benchmarks were performed with this [software](https://crystalmark.info/en/software/crystaldiskmark/). SSD hit 1.5 G/s write times, which is about an order of magnitude faster than the HD. Problem solved.

Should we need to put a few larger SSDs in some of these computers, that is not a problem at all. Given the low capacity of the SSD (1 TB), we will have to be diligent to upload data to BioHPC upon completion.  I would also be curious to see if we can create a direct mount to BioHPC and upload directly. Should be by far the fastest...

closed 2023-08-23 by AdvancedImagingUTSW