TheDeanLab/navigate #602: Lasers must be defined in all microscopes in `configuration.yaml` in order to be used

zacsimile, 2023-08-23:
Tested on the upright. If we have a laser defined in the second microscope in `configuration.yaml`, but not the first, it will appear in the laser dropdown list for the second microscope only in the GUI. This is good. However, when this unique laser is selected, the last laser to be defined in the first microscope will turn on instead of the selected laser. If the laser is defined in both systems, it works as expected.

AdvancedImagingUTSW, 2025-02-12:
@annie-xd-wang - I would be curious if this was addressed during the device refactor that you recently did.

annie-xd-wang, 2025-02-13:
This issue is not related to how we define or load the lasers; it's more of a controller-side issue. I've submitted a [PR 1061](https://github.com/TheDeanLab/navigate/pull/1061) to fix it.

closed 2025-02-13 by AdvancedImagingUTSW