TheDeanLab/navigate #604: Turn off lasers whenever data thread is paused

zacsimile, 2023-08-23:
If the data thread is paused, the signal thread can continue moving stages but must turn off the laser until the data thread is resumed. This will prevent our samples from bleaching during long pauses.

AdvancedImagingUTSW, 2023-08-23:
I feel like a better solution would be to toggle the lasers on immediately prior to an acquisition, and toggle them off immediately after, in all circumstances.

We turn the lasers on within the `microscope` object, `prepare_next_channel` method. To turn them off, we would just need to do the following...
```self.lasers[str(self.laser_wavelength[current_laser_index])].turn_off()```

AdvancedImagingUTSW, 2023-08-23:
Made a quick push to linked branch that explicitly turns off the lasers after each image acquisition...

https://github.com/TheDeanLab/ASLM/tree/604-turn-off-lasers-whenever-data-thread-is-paused

zacsimile, 2023-08-23:
Yeah this is a good idea. I think keeping the laser on in between acquisitions is what's causing this central stripe. If you look at the galvo waveforms, they are parking the beam in the center of the frame in between acquisitions. 

[image]

[image]

zacsimile, 2023-08-23:
Can you PR this so we can test?

AdvancedImagingUTSW, 2023-08-23:
That central stripe definitely appears to be a parked beam...  Hopefully this is sufficient to eliminate that.

closed 2023-08-27 by zacsimile