TheDeanLab/navigate #621: can't truncate width of lightning camera in camera view live mode

mehr0096, 2023-09-18:
Get error when trying to truncate FOV in x direction (width). We are able to truncate y direction (height)
```
camera thread exception happened!  Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 317, in _recv_bytes
    nread, err = ov.GetOverlappedResult(True)
BrokenPipeError: [WinError 109] The pipe has been ended

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 175, in func
    target(*args, **kwargs)
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\controller.py", line 815, in capture_image
    image_id = self.show_img_pipe.recv()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 255, in recv
    buf = self._recv_bytes()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 326, in _recv_bytes
    raise EOFError
EOFError

camera thread ended because of exception!: not all arguments converted during string formatting Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 317, in _recv_bytes
    nread, err = ov.GetOverlappedResult(True)
BrokenPipeError: [WinError 109] The pipe has been ended

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 175, in func
    target(*args, **kwargs)
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\controller.py", line 815, in capture_image
    image_id = self.show_img_pipe.recv()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 255, in recv
    buf = self._recv_bytes()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 326, in _recv_bytes
    raise EOFError
EOFError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 64, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 183, in func
    logger.debug(
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "c:\users\dean-lab\documents\aslm\src\aslm\log_files\filters.py", line 81, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

Exception in thread camera:
Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 317, in _recv_bytes
    nread, err = ov.GetOverlappedResult(True)
BrokenPipeError: [WinError 109] The pipe has been ended

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 175, in func
    target(*args, **kwargs)
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\controller.py", line 815, in capture_image
    image_id = self.show_img_pipe.recv()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 255, in recv
    buf = self._recv_bytes()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\multiprocessing\connection.py", line 326, in _recv_bytes
    raise EOFError
EOFError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 64, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 183, in func
    logger.debug(
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "c:\users\dean-lab\documents\aslm\src\aslm\log_files\filters.py", line 81, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "c:\users\dean-lab\documents\aslm\src\aslm\controller\thread_pool.py", line 67, in run
    logger.debug(f"{self.name} thread ended because of exception!: {e}", traceback.format_exc())
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "c:\users\dean-lab\documents\aslm\src\aslm\log_files\filters.py", line 81, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

```

mehr0096, 2023-09-19:
[model_debug.log](https://github.com/TheDeanLab/ASLM/files/12664272/model_debug.log)
[view_controller_debug.log](https://github.com/TheDeanLab/ASLM/files/12664273/view_controller_debug.log)
[view_controller_info.log](https://github.com/TheDeanLab/ASLM/files/12664274/view_controller_info.log)

zacsimile, 2023-09-19:
Are these the log files from this session? It doesn't look like you try to set the width lower in the log files.

mehr0096, 2023-09-19:
For ORCA lightning, can only shrink by a multiple of 4 from 4608.

closed 2023-09-19 by mehr0096