TheDeanLab/navigate #630: Camera Exposure/Line Interval Error

AdvancedImagingUTSW, 2023-09-27:
Will follow up with a traceback tomorrow, but this line of code in `microscope.prepare_next_channel()` is throwing an error...

```
# Camera Settings
        self.current_exposure_time = float(channel["camera_exposure_time"])
        if (
            self.configuration["experiment"]["CameraParameters"]["sensor_mode"]
            == "Light-Sheet"
        ):
            (
                self.current_exposure_time,
                camera_line_interval,
            ) = self.camera.calculate_light_sheet_exposure_time(
                self.current_exposure_time,
                int(
                    self.configuration["experiment"]["CameraParameters"][
                        "number_of_pixels"
                    ]
                ),
            )
            self.camera.set_line_interval(camera_line_interval)
        self.camera.set_exposure_time(self.current_exposure_time)
```

mehr0096, 2023-09-27:
```exposure_time Configuration Failed 0.00743218 0.0075806400000000005
Traceback (most recent call last):
  File "c:\users\dean-lab\documents\aslm\src\aslm\model\devices\daq\daq_ni.py", line 497, in stop_acquisition
    self.camera_trigger_task.stop()
AttributeError: 'NoneType' object has no attribute 'stop'```

zacsimile, 2023-09-27:
@mehr0096 I think this is a different error, related to the DAQ rather than the camera. The error @AdvancedImagingUTSW mentioned has to do with a mismatch in the set exposure time/line interval.

AdvancedImagingUTSW, 2023-09-30:
I was the one who posted that wrong traceback previously. Not sure what I was thinking at the time.

Looking into this issue.  Presumably, this issue is happening for all of our Hamamatsu Cameras. Currently debugging on the Lightning, but confirmed it also happens on the BT-MesoSPIM. It happens when we set the exposure time. According to the documentation, the step size for incrementing the exposure time is quite small. 

[image]

For the settings on the software, 100 ms exposure and 100 pixel rolling shutter readout, the exposure time passed to the `set_property_value` method in the `HamamatsuAPI` are the following is 0.00465766.  

The API returns a minimum value of 6.304e-6, maximum value of 0.0002, and a step size of 6.4e-9. However, when we set the value, the exposure time is actually 0.0004843904. This value is within ~4% of what we want, but we set an arbitrary value for throwing the error at 1%. Nonetheless, I'm not sure why the returned value is so much different than what we would expect since the step size is actually quite small.  A poor fix is to increase our arbitrary threshold for triggering the print statement:

```
        if abs(final_configuration - value) < value / 100:
            return True
        else:
            print(name, "Configuration Failed", value, final_configuration)
            return False
```

AdvancedImagingUTSW, 2023-09-30:
If I change the arbitrary threshold to 5%, it does not throw the error, but that feels pretty lame. 

If I go to where we are actually doing the work, in `prop_setgetvalue`:

```        cDouble = c_double(fValue)
        cOption = c_int32(option)
        cIDProp = c_int32(idprop)
        ret = self.__result(
            dcamprop_setgetvalue(self.__hdcam, cIDProp, byref(cDouble), cOption)
        )
        if ret is False:
            return False

        return cDouble.value
```
Looking at a few other repos, this looks pretty standard...  Not sure.

zacsimile, 2023-10-09:
This is the error:

`exposure_time Configuration Failed 0.0051404 0.005235545112781955`

AdvancedImagingUTSW, 2023-10-09:
I pushed one change to this branch where I set the threshold for throwing the error to 5% instead of 1%, which is what it is now. 

What doesn't necessarily make sense to me is why the difference between the set value and the actual value are so different. The increment for the exposure time is very small for these cameras (e.g., sub-microsecond).

zacsimile, 2023-11-04:
This is not as easy as rounding to the nearest step size. Something else is going on. I suspect it may have to do with the order of calling the update to the line interval vs. the exposure time. Playing around with modifying the order of these things in HCImage is probably the easiest way to figure it out.

AdvancedImagingUTSW, 2024-05-10:
This issue was tackled by @annie-xd-wang several months ago. It involved the camera model specific calculation of the maximum frame rate. Both the `H` value, and the `Calculation Formula`, vary according to the camera model. 

We now have multiple Hamamatsu classes, each with their own calculation. These include the Orca Flash 4.0, Orca Fusion, Orca Lightning, and Orca Fire. In the future, we could also consider adding the Quest.

[image]

closed 2024-05-10 by AdvancedImagingUTSW