TheDeanLab/navigate #641: ASI stage fails silently on incorrect `axes_mapping`

zacsimile, 2023-10-19:
If we do a lowercase `axes_mapping`, e.g. `[x, y, z]` instead of `[X, Y, Z]`, the ASI stage does not work properly, but fails to alert the user that they have provided non-existent axes in the axis mapping. We should check that the axes provided in `axes_mapping` exist within the available set of axes on the Tiger controller.

AdvancedImagingUTSW, 2023-10-19:
I literally just ran into this error yesterday helping out the University of Colorado team that is building an OPM. 

Would be valuable to make sure that the ASI stages are capitalized, which should largely fix it (so long as that axis exists)

jondaniels, 2023-10-20:
The ASI stage should respond identically to upper and lower case characters; the first thing the parser does is convert everything to uppercase.  If it does not I would appreciate knowing how to replicate the error.  Could the error be on the Python side?

zacsimile, 2023-10-23:
@AdvancedImagingUTSW, yep, that fixes it.

@jondaniels No error on the ASI side. See #644.

PR #644 (2023-10-24): Force ASI axes to uppercase
Closes #641. The problem is in `get_position()` in  `asi_tiger_controller.py`:

```
axes_seq = list(
    filter(
        lambda axis: axis if axis in axes else False,
        self.default_axes_sequence,
    )
)
```

`self.default_axes_sequence` is all upper case when it is pulled from the stage, so the lower case values aren't passed and no position is reported. However, just fixing this without forcing `asi_axes` to upper case in `stage_asi.py` results in key errors. Easiest to just force the axes all to upper case when they are read in.
closed 2023-10-24 by AdvancedImagingUTSW