TheDeanLab/navigate #657: Cleanup Requirements.txt and/or replace with pyproject.toml

AdvancedImagingUTSW, 2023-10-30:
I'm pretty sure we no longer need tensor flow in the requirements, which would clean up the dependency install substantially. Could do this while updating to a pyproject.toml, if we think that is a good idea.

Also, the Capstone team will need this library: https://github.com/Mecademic/mecademicpy Not sure if we should add this to our requirements, add it as a sub-module, or install a static version in our API folder. Open for opinions.

zacsimile, 2023-11-04:
I'm in favor of switching to `pyproject.toml` and having like a `robot` dependency set for the mecademic.

AdvancedImagingUTSW, 2023-11-06:
Started some work on this on the linked branch. The pyproject.toml is defined and appears to work for the main application. Sphinx fails currently when I try `make html`

AdvancedImagingUTSW, 2023-11-06:
@conorhughmcfadden coild you list any dependencies you need for the AO in addition to what is standard?

closed 2023-11-09 by zacsimile