TheDeanLab/navigate #671: Analog Output Error

EvoleneP, 2023-11-15:
**Describe the bug**
When the software is configured to use the NI data acquisition card, but there are no analog output signals configured (all set as Synthetic variants), runs into a problem with the `self.n_sample = min(n_samples)` command in `create_analog_output_tasks` given that `n_sampes` is equal to 0.

**Traceback**
```
Traceback (most recent call last):
  File "C:\Users\Nikon\miniconda3\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Nikon\miniconda3\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\nikon\documents\github\aslm\src\aslm\model\model.py", line 985, in run_live_acquisition
    self.run_acquisition()
  File "c:\users\nikon\documents\github\aslm\src\aslm\model\model.py", line 1003, in run_acquisition
    self.snap_image()
  File "c:\users\nikon\documents\github\aslm\src\aslm\model\model.py", line 963, in snap_image
    self.active_microscope.daq.prepare_acquisition(
  File "c:\users\nikon\documents\github\aslm\src\aslm\model\devices\daq\daq_ni.py", line 335, in prepare_acquisition
    self.create_analog_output_tasks(channel_key)
  File "c:\users\nikon\documents\github\aslm\src\aslm\model\devices\daq\daq_ni.py", line 250, in create_analog_output_tasks
    self.n_sample = min(n_samples)
ValueError: min() arg is an empty sequence
```
**Logs & Configuration Files**

zacsimile, 2023-11-15:
Thanks for reporting this @EvoleneP! We have not tried to run the software in this scenario yet. If you are comfortable with it, you migh try updating 

```
def create_analog_output_tasks(self, channel_key):
        """Create analog output tasks for each board.

        Create a single analog output task for all channels per board. Most NI DAQ cards
        have only one clock for analog output sample timing, and as such all channels
        must be grouped here.

        Parameters
        ----------
        channel_key : str
            Channel key for analog output.
        """
        n_samples = list(set([v["samples"] for v in self.analog_outputs.values()]))
        if len(n_samples) > 1:
            logger.debug(
                "NI DAQ - Different number of samples provided for each analog"
                "channel. Defaulting to the minimum number of samples provided."
                "Waveforms will be clipped to this length."
            )
```

to

```
def create_analog_output_tasks(self, channel_key):
        """Create analog output tasks for each board.

        Create a single analog output task for all channels per board. Most NI DAQ cards
        have only one clock for analog output sample timing, and as such all channels
        must be grouped here.

        Parameters
        ----------
        channel_key : str
            Channel key for analog output.
        """
        n_samples = list(set([v["samples"] for v in self.analog_outputs.values()]))
        if len(n_samples) == 0:
            return
        elif len(n_samples) > 1:
            logger.debug(
                "NI DAQ - Different number of samples provided for each analog"
                "channel. Defaulting to the minimum number of samples provided."
                "Waveforms will be clipped to this length."
            )
```

If you are uncomfortable, I can have a go at coding this and make a PR branch for you to try. Incidentally, `self.n_samples` shouldn't be a class attribute, I think, as we access it nowhere outside of this function.

AdvancedImagingUTSW, 2023-11-15:
`self.n_sample` is also called by `update_analog_task` in `daq_ni`, so I'll leave it for now. Was tempted to try to make a test, but it would be a significant effort. I helped her test this change on her machine, and it worked. I'm okay with merging.

zacsimile, 2023-11-15:
Ah you're right. Let's leave it.

closed 2023-11-16 by AdvancedImagingUTSW