TheDeanLab/navigate #674: Z-stack error

zacsimile, 2023-11-15:
```
Exception in thread z-stack Data:
Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Dean-Lab\miniconda3\envs\ASLM\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\model.py", line 806, in run_data_process
    data_func(frame_ids)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\features\image_writer.py", line 254, in save_image
    image = self.data_buffer[idx][:, ::-1]
IndexError: list index out of range
```

zacsimile, 2023-11-16:
I'm gonna go ahead and say it's fine to crash on this. This shouldn't happen, and the user should be alerted.

zacsimile, 2023-11-16:
I changed my mind. We shouldn't allow this.

AdvancedImagingUTSW, 2023-11-16:
I assume that this is unrelated to the extra frame we are getting in with the `NIGalvo`-based z-stacking... 

It also seems like the log files would provide some insight into the problem...

zacsimile, 2023-11-16:
It's unrelated to the galvo z-stacking, happened on the mesoSPIM BT. I PRed a fix.

AdvancedImagingUTSW, 2023-11-16:
```me Configuration Failed 0.0025702 0.0026176691729323308
Coupled axes: None <class 'NoneType'>
exposure_time Configuration Failed 0.0025702 0.0026176691729323308
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
error message:  DCAMERR.TIMEOUT
Received invalid index -94. Skipping this frame.
Received invalid index -93. Skipping this frame.
Received invalid index -92. Skipping this frame.
Received invalid index -91. Skipping this frame.
Received invalid index -90. Skipping this frame.
Received invalid index -89. Skipping this frame.
Received invalid index -88. Skipping this frame.
Received invalid index -87. Skipping this frame.
Received invalid index -86. Skipping this frame.
Received invalid index -85. Skipping this frame.
Received invalid index -84. Skipping this frame.
Received invalid index -83. Skipping this frame.
Received invalid index -82. Skipping this frame.
Received invalid index -81. Skipping this frame.
Received invalid index -80. Skipping this frame.
Received invalid index -79. Skipping this frame.
Received invalid index -78. Skipping this frame.
Received invalid index -77. Skipping this frame.
Received invalid index -76. Skipping this frame.
Received invalid index -75. Skipping this frame.
Received invalid index -74. Skipping this frame.
Received invalid index -73. Skipping this frame.
Received invalid index -72. Skipping this frame.
Received invalid index -71. Skipping this frame.
Received invalid index -70. Skipping this frame.
Received invalid index -69. Skipping this frame.
Received invalid index -68. Skipping this frame.
Received invalid index -67. Skipping this frame.
Received invalid index -66. Skipping this frame.
Received invalid index -65. Skipping this frame.
Received invalid index -64. Skipping this frame.
Received invalid index -63. Skipping this frame.
Received invalid index -62. Skipping this frame.
Received invalid index -61. Skipping this frame.
Received invalid index -60. Skipping this frame.
Received invalid index -59. Skipping this frame.
Received invalid index -58. Skipping this frame.
Received invalid index -57. Skipping this frame.
Received invalid index -56. Skipping this frame.
Received invalid index -55. Skipping this frame.
Received invalid index -54. Skipping this frame.
Received invalid index -53. Skipping this frame.
Received invalid index -52. Skipping this frame.
Received invalid index -51. Skipping this frame.
Received invalid index -50. Skipping this frame.
Received invalid index -49. Skipping this frame.
Received invalid index -48. Skipping this frame.
Received invalid index -47. Skipping this frame.
Received invalid index -46. Skipping this frame.
Received invalid index -45. Skipping this frame.
Received invalid index -44. Skipping this frame.
Received invalid index -43. Skipping this frame.
Received invalid index -42. Skipping this frame.
Received invalid index -41. Skipping this frame.
Received invalid index -40. Skipping this frame.
Received invalid index -39. Skipping this frame.
Received invalid index -38. Skipping this frame.
Received invalid index -37. Skipping this frame.
Received invalid index -36. Skipping this frame.
Received invalid index -35. Skipping this frame.
Received invalid index -34. Skipping this frame.
Received invalid index -33. Skipping this frame.
Received invalid index -32. Skipping this frame.
Received invalid index -31. Skipping this frame.
Received invalid index -30. Skipping this frame.
Received invalid index -29. Skipping this frame.
Received invalid index -28. Skipping this frame.
Received invalid index -27. Skipping this frame.
Received invalid index -26. Skipping this frame.
Received invalid index -25. Skipping this frame.
Received invalid index -24. Skipping this frame.
Received invalid index -23. Skipping this frame.
Received invalid index -22. Skipping this frame.
Received invalid index -21. Skipping this frame.
Received invalid index -20. Skipping this frame.
Received invalid index -19. Skipping this frame.
Received invalid index -18. Skipping this frame.
Received invalid index -17. Skipping this frame.
Received invalid index -16. Skipping this frame.
Received invalid index -15. Skipping this frame.
Received invalid index -14. Skipping this frame.
Received invalid index -13. Skipping this frame.
Received invalid index -12. Skipping this frame.
Received invalid index -11. Skipping this frame.
Received invalid index -10. Skipping this frame.
Received invalid index -9. Skipping this frame.
Received invalid index -8. Skipping this frame.
Received invalid index -7. Skipping this frame.
Received invalid index -6. Skipping this frame.
Received invalid index -5. Skipping this frame.
Received invalid index -4. Skipping this frame.
Received invalid index -3. Skipping this frame.
Received invalid index -2. Skipping this frame.
Received invalid index -1. Skipping this frame.

```

zacsimile, 2023-11-16:
The fix is behaving correctly. We should not receive negative frame indices. What were the steps to get to this point?

zacsimile, 2023-11-16:
OK, so, looking at the logs, the data thread didn't run (was it paused?) for 253 frames. This will kick us into the `else` case in `aslm.model.devices.APIs.hamamatsu.HamamatsuAPI.get_frames()`. Since our data buffer size is only `100` and we waited `253` frames, this generated negative indices. To solve this, we need to figure out why the data thread was paused for so long.

zacsimile, 2023-11-16:
This appears to be an issue when saving takes a long time. It does not appear to be an issue with pausing the data buffer.

PR #676 (2023-11-16): Skip unusable indices in `save_image()`
Closes #674.
closed 2023-11-22 by zacsimile