TheDeanLab/navigate #677: Model called twice

zacsimile, 2023-11-16:
It's time to add a `_Custody` object to prevent these kind of errors.

```
model thread exception happened! Two different threads tried to use the same ObjectInSubprocess at the same time! This is bad. Look at the docstring of concurrency_tools.py to see an example of how to use a _Custody object to avoid this problem. Traceback (most recent call last):
  File "c:\users\dean-lab\desktop\aslm\src\aslm\controller\thread_pool.py", line 251, in func
    target(*args, **kwargs)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\controller\controller.py", line 641, in <lambda>
    "model", lambda: self.model.run_command("update_setting", *args)
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\concurrency\concurrency_tools.py", line 404, in __getattr__
    with self._.parent_pipe_lock:
  File "c:\users\dean-lab\desktop\aslm\src\aslm\model\concurrency\concurrency_tools.py", line 578, in __enter__
    raise RuntimeError(
RuntimeError: Two different threads tried to use the same ObjectInSubprocess at the same time! This is bad. Look at the docstring of concurrency_tools.py to see an example of how to use a _Custody object to avoid this problem.
```

zacsimile, 2023-11-17:
Further investigation revels we are keeping track of which threads are accessing the model in most cases. There are only a few places where it is possible to access the model from two different threads simultaneously. We should keep an eye out for what tasks we are performing when this error arises.

annie-xd-wang, 2023-12-06:
The start-and-stop acquisition is fine since the Acquire/Stop button blocks unsafe calling. All calls to the model are put into the thread pool now, this error shouldn't happen anymore.

closed 2023-12-06 by zacsimile