TheDeanLab/navigate #709: Handle if we are in live acquisition mode in Autofocus

zacsimile, 2023-12-06:
Right now we have to stop live acquisition mode to launch autofocus. Make it so we automatically stop live acquisition mode, run autofocus, restart live acquisition mode if this is the case.

AdvancedImagingUTSW, 2023-12-27:
@annie-xd-wang, this is close to functional, but not. Any chance you could look at the branch that is linked to this? Essentially, we now pass if we should return to the live acquisition mode as a kwarg, but after running the autofocus routine (`autofocus.run`), I'm not sure what the best way to create a wait statement for the autofocus to complete. I assume it may be one or two lines of code, but everything I have tried doesn't work as intended because the autofocus routine is launched as a asynchronous thread.

annie-xd-wang, 2023-12-27:
Sure, I will look into it.

AdvancedImagingUTSW, 2024-02-02:
This was fixed in #799

closed 2024-02-02 by AdvancedImagingUTSW