TheDeanLab/navigate #728: Exposure time not updated properly.

AdvancedImagingUTSW, 2023-12-14:
**Describe the bug**
Strange error, perhaps related to one that @JinlongL reported a while back with the filter wheel. It seems as if the software doesn't grab the most recent state of the channel settings. For example, we cranked up the exposure time for the far-red channel here, but when we started the exposure, the green channel also has this exposure. 

[image]

[image]

zacsimile, 2023-12-14:
So the Framerate info is updated independently of the rest of the software in `channels_tab_controller.execute()`. We can see on line 877 that "Exposure time (ms)" is set to the maximum exposure time over all channels. I don't think this is propagated elsewhere, but it's worth checking.

annie-xd-wang, 2023-12-14:
This bug is not caused by the controller side. It's because the sample counts passed to the DAQ are wrong. I'm fixing it.

closed 2023-12-14 by zacsimile