TheDeanLab/navigate #732: Camera trigger task not cleaned up properly

mehr0096, 2023-12-14:
```
Traceback (most recent call last):
  File "C:\Users\Dean-Lab\miniconda3\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Dean-Lab\miniconda3\envs\navigate\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Dean-Lab\Documents\ASLM\src\navigate\model\model.py", line 1034, in run_live_acquisition
    self.run_acquisition()
  File "C:\Users\Dean-Lab\Documents\ASLM\src\navigate\model\model.py", line 1052, in run_acquisition
    self.snap_image()
  File "C:\Users\Dean-Lab\Documents\ASLM\src\navigate\model\model.py", line 1013, in snap_image
    self.active_microscope.daq.prepare_acquisition(
  File "C:\Users\Dean-Lab\Documents\ASLM\src\navigate\model\devices\daq\daq_ni.py", line 343, in prepare_acquisition
    self.create_camera_task(channel_key)
  File "C:\Users\Dean-Lab\Documents\ASLM\src\navigate\model\devices\daq\daq_ni.py", line 228, in create_camera_task
    self.camera_trigger_task.co_channels.add_co_pulse_chan_time(
  File "C:\Users\Dean-Lab\miniconda3\envs\navigate\lib\site-packages\nidaqmx\_task_modules\co_channel_collection.py", line 208, in add_co_pulse_chan_time
    check_for_error(error_code)
  File "C:\Users\Dean-Lab\miniconda3\envs\navigate\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Specified channel cannot be added to the task, because a channel with the same name is already in the task.
Channel Name: PXI6259/ctr0
```
[view_controller_debug.log](https://github.com/TheDeanLab/navigate/files/13679271/view_controller_debug.log)
[model_debug.log](https://github.com/TheDeanLab/navigate/files/13679272/model_debug.log)

zacsimile, 2023-12-26:
Can you attach the configuration file as well? I assume this means the camera task isn't closed out and a fresh camera task gets generated prematurely? Is this still happening?

AdvancedImagingUTSW, 2024-03-04:
I'm starting to believe that this is no longer an issue and can be closed. Any opinion @mehr0096?

mehr0096, 2024-03-04:
This issue can be closed.

closed 2024-03-04 by AdvancedImagingUTSW