TheDeanLab/navigate #736: ExASLM_Stage_Z_Stack_Issue

JinlongL, 2023-12-17:
When acquiring Z stack with ASI FTP2000 stage, some noticeable lines at the beginning and the middle of MIP of axial planes (xz, yz). 

This issue is fixed with backing the stage back to the starting position and move towards the start position 20 um. Probably some lead time is needed for the stage to get ready for each Z stack acquisition.
No backup:
[image]
With backup:
[image]

zacsimile, 2023-12-26:
What is the path forward here? Do we want to add a `lead_distance` option for ASI stages that automatically backs it off its starting position by `lead_distance` whenever a ZStack (or other non-live acquisition) is started?

JinlongL, 2023-12-26:
Hi Zach,

Thank you for checking on this. I think adding some lead time could be the best option here until they fix the bug.

Thanks,
Jinlong

EXTERNAL MAIL

CAUTION: This email originated from outside UTSW. Please be cautious of links or attachments, and validate the sender's email address before replying.

UT Southwestern

Medical Center

The future of medicine, today.

AdvancedImagingUTSW, 2024-06-03:
Presumably fixed with hardware changes. Can reactivate if this resurfaces.

closed 2024-06-03 by AdvancedImagingUTSW