TheDeanLab/navigate #743: KeyError - 'channel_0'

AdvancedImagingUTSW, 2023-12-20:
**Describe the bug**
While imaging on Voodoo, seem to run into the following error. Only CH3 was selected in the GUI.

**Traceback**
```

(navigate) C:\Users\Colfax4>navigate
Exception in thread live signal:
Traceback (most recent call last):
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\model.py", line 1010, in snap_image
    self.active_microscope.daq.run_acquisition()
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 384, in run_acquisition
    if self.camera_trigger_task.is_task_done():
  File "C:\Users\Colfax4\.conda\envs\navigate\lib\site-packages\nidaqmx\task.py", line 520, in is_task_done
    check_for_error(error_code)
  File "C:\Users\Colfax4\.conda\envs\navigate\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Task specified is invalid or does not exist.
Status Code: -200088

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Colfax4\.conda\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Colfax4\.conda\envs\navigate\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\model.py", line 1034, in run_live_acquisition
    self.run_acquisition()
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\model.py", line 1052, in run_acquisition
    self.snap_image()
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\model.py", line 1013, in snap_image
    self.active_microscope.daq.prepare_acquisition(
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 362, in prepare_acquisition
    self.create_camera_task(channel_key)
  File "C:\Users\Colfax4\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 223, in create_camera_task
    exposure_time = self.exposure_times[channel_key]
KeyError: 'channel_0'
exposure_time Configuration Failed 0.00376647 0.00395478947368421
```

zacsimile, 2023-12-20:
Can you attach the logs?

AdvancedImagingUTSW, 2024-01-04:
This one just popped up on CT-ASLM-V2. 

```
(navigate) C:\Users\Spectral>navigate
Exception in thread live signal:
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 1051, in snap_image
    self.active_microscope.daq.run_acquisition()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 365, in run_acquisition
    if self.camera_trigger_task.is_task_done():
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\nidaqmx\task.py", line 520, in is_task_done
    check_for_error(error_code)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Task specified is invalid or does not exist.
Status Code: -200088

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 1075, in run_live_acquisition
    self.run_acquisition()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 1093, in run_acquisition
    self.snap_image()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 1054, in snap_image
    self.active_microscope.daq.prepare_acquisition(
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 343, in prepare_acquisition
    self.create_camera_task(channel_key)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\daq\daq_ni.py", line 223, in create_camera_task
    exposure_time = self.exposure_times[channel_key]
KeyError: 'channel_0'
```
When it occurred here, I was trying to update the stage position by repetitively hitting the stop button on the stage tab while acquiring in the continuous mode. 

```
2024-01-04 15:10:34,677 - model - INFO - model: Navigate Model - Received frames in total: 121
2024-01-04 15:10:34,719 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -954201.88, 'x_pos': 36685.6, 'f_pos': 2162697.92}
2024-01-04 15:10:34,720 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': 2141723.08, 'x_pos': 33318499.32, 'f_pos': 128457.68000000001}
2024-01-04 15:10:34,767 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': 32884531.2, 'x_pos': 9395227.64, 'f_pos': 149422.6}
2024-01-04 15:10:34,989 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-04 15:10:34,990 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-04 15:10:35,008 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': 32885033.48, 'x_pos': -3404.0, 'f_pos': 2162694.6}
2024-01-04 15:10:35,055 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': 304087.56, 'x_pos': -3404.0, 'f_pos': 2162692.64}
2024-01-04 15:10:35,061 - model - INFO - daq_ni: Waveform Template Name: Default
2024-01-04 15:10:35,067 - model - INFO - daq_ni: Waveform Expand Num = 1
2024-01-04 15:10:35,068 - model - INFO - daq_ni: Waveform Repeat Num = 1
2024-01-04 15:12:26,115 - model - INFO - model: Navigate Model - Stopping with stop command.
2024-01-04 15:12:26,164 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -3404.0, 'x_pos': 8368.36, 'f_pos': 1694.24}
2024-01-04 15:12:26,196 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -3404.0, 'x_pos': 8368.36, 'f_pos': 1694.24}
2024-01-04 15:12:29,082 - model - DEBUG - stage_sutter: MP-285 stage connection closed
2024-01-04 15:12:29,172 - model - INFO - camera_hamamatsu: HamamatsuOrca Shutdown
2024-01-04 15:12:29,172 - model - DEBUG - remote_focus_equipment_solutions: Closing RemoteFocusEquipmentSolutions Serial Port
2024-01-04 15:12:29,172 - model - DEBUG - zoom_synthetic: SyntheticZoom Servo instance Deleted
2024-01-04 15:12:29,172 - model - ERROR - stage_sutter: MP-285 - Stage stop failed: ClearCommError failed (OSError(9, 'The handle is invalid.', None, 6))
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\stages\stage_sutter.py", line 272, in stop
    self.stage.interrupt_move()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 383, in interrupt_move
    self.safe_write(bytes.fromhex("03"))
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 101, in safe_write
    self.serial.read_all()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialutil.py", line 652, in read_all
    return self.read(self.in_waiting)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (OSError(9, 'The handle is invalid.', None, 6))
2024-01-04 15:12:29,172 - model - DEBUG - stage_sutter: MP-285 stage connection closed
```

closed 2024-01-08 by AdvancedImagingUTSW