TheDeanLab/navigate #761: Run Autofocus from Menu

AdvancedImagingUTSW, 2023-12-26:
**Describe the bug**
Occurs when one runs the autofocus routine, but from the menu item, and not the popup.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the software
2. Select Autofocus -> Perform Autofocus.

**Expected behavior**
It should presumably use the default stage. Problem now occurs because we can perform the operation on any stage that we want, and this was implemented before that.

**Traceback**
```Exception in Tkinter callback
Traceback (most recent call last):
  File "/opt/miniconda3/envs/navigate/lib/python3.9/tkinter/__init__.py", line 1892, in __call__
    return self.func(*args)
TypeError: <lambda>() missing 1 required positional argument: 'x'
```

zacsimile, 2023-12-26:
Needs a `device` and a `device_ref` (see line 133 of `autofocus_popup_controller.py`).

AdvancedImagingUTSW, 2023-12-26:
I’m looking into this. I may have created it while working on #709

AdvancedImagingUTSW, 2023-12-27:
False positive. Closing.

closed 2023-12-27 by AdvancedImagingUTSW