TheDeanLab/navigate #774: DevicePluginExample Throws Error

AdvancedImagingUTSW, 2024-01-02:
While following the installation instructions we have on the website (https://thedeanlab.github.io/navigate/quick_start.html) I ran into the following error:

```
(base) C:\Users\Kevin>conda activate navigate

(navigate) C:\Users\Kevin>navigate -sh
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
vvvvvvvvvvvvvvvvvvvvvvvvvvv Child Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvv
 Traceback (most recent call last):
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 501, in _child_loop
    obj = initializer(*initargs, **initkwargs)
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\model\model.py", line 115, in __init__
    plugin_devices, plugin_acquisition_modes = plugins.load_plugins()
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\model\plugins_model.py", line 145, in load_plugins
    module = load_module_from_file(
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\tools\common_functions.py", line 124, in load_module_from_file
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 846, in exec_module
  File "<frozen importlib._bootstrap_external>", line 982, in get_code
  File "<frozen importlib._bootstrap_external>", line 1039, in get_data
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Kevin\\.conda\\envs\\navigate\\Lib\\site-packages\\navigate\\plugins\\DevicePluginExample\\model\\devices\\__pycache__\\device_startup_functions.py'
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^ Child Process Traceback ^^^^^^^^^^^^^^^^^^^^^^^^^^^
 vvvvvvvvvvvvvvvvvvvvvvvvvvv Main Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvvv
Traceback (most recent call last):
  File "C:\Users\Kevin\.conda\envs\navigate\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Kevin\.conda\envs\navigate\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Kevin\.conda\envs\navigate\Scripts\navigate.exe\__main__.py", line 7, in <module>
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\main.py", line 105, in main
    Controller(
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\controller\controller.py", line 160, in __init__
    self.model = ObjectInSubprocess(
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 383, in __init__
    assert _get_response(self) == "Successfully initialized"
  File "C:\Users\Kevin\.conda\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Kevin\\.conda\\envs\\navigate\\Lib\\site-packages\\navigate\\plugins\\DevicePluginExample\\model\\devices\\__pycache__\\device_startup_functions.py'

(navigate) C:\Users\Kevin>
```

annie-xd-wang, 2024-01-02:
This error is solved in [PR#767](https://github.com/TheDeanLab/navigate/pull/767)

AdvancedImagingUTSW, 2024-01-04:
I can confirm that #767 eliminates this issue. Closing.

closed 2024-01-04 by AdvancedImagingUTSW