TheDeanLab/navigate #780: Waveform Settings Tab

AdvancedImagingUTSW, 2024-01-04:
```
(navigate) C:\Users\Spectral>navigate
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\waveform_tab_controller.py", line 217, in plot_waveforms
    max_remote_focus_waveform = np.maximum(max_remote_focus_waveform, np.max(remote_focus_waveform))
TypeError: '>=' not supported between instances of 'int' and 'NoneType'
```
I suspect it results from the different exposure times but I have not thought about it much.

[image]

zacsimile, 2024-01-06:
Could you upload the log files? Was this before or after pressing "Acquire" and then "Stop"?

AdvancedImagingUTSW, 2024-01-11:
We were using the software as normal. Acquired data in a multi-format mode. Everything went well. Selected the waveform tab and got the following error in the Anaconda prompt.

```
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\waveform_tab_controller.py", line 217, in plot_waveforms
    max_remote_focus_waveform = np.maximum(max_remote_focus_waveform, np.max(remote_focus_waveform))
TypeError: '>=' not supported between instances of 'float' and 'NoneType'
```

Last few lines from model.debug
<details>

```
2023-12-12 19:10:36,369 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -8973.6, 'x_pos': 8578.56, 'f_pos': 2553.2400000000002}
2023-12-12 19:10:36,578 - model - INFO - model: Navigate Model - Running data process, get frames [99]
2023-12-12 19:10:36,617 - model - INFO - model: Navigate Model - Sent through pipe99
2023-12-12 19:10:36,718 - model - DEBUG - feature_container: running signal node: ZStackAcquisition
2023-12-12 19:10:36,724 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Moving to Position 0
2023-12-12 19:10:36,842 - model - DEBUG - feature_container: running signal node: StackPause
2023-12-12 19:10:36,842 - model - DEBUG - feature_container: running signal node: LoopByCount
2023-12-12 19:10:37,118 - model - DEBUG - HamamatsuAPI: error message: -2147483386
2023-12-12 19:10:37,118 - model - INFO - model: Navigate Model - Running data process, get frames []
2023-12-12 19:10:37,118 - model - INFO - model: Navigate Model - Waiting 20
2023-12-12 19:10:37,118 - model - INFO - model: Navigate Model - Data thread stopped.
2023-12-12 19:10:37,118 - model - INFO - model: Navigate Model - Received frames in total: 3000
2023-12-12 19:10:37,168 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -8973.6, 'x_pos': 8578.56, 'f_pos': 2553.2400000000002}
2023-12-12 19:10:50,018 - model - INFO - model: Navigate Model - Stopping with stop command.
2023-12-12 19:10:50,067 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -8973.6, 'x_pos': 8578.56, 'f_pos': 2553.2400000000002}
2023-12-12 19:10:50,116 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -8973.6, 'x_pos': 8578.56, 'f_pos': 2553.2400000000002}
2023-12-12 19:10:53,118 - model - DEBUG - stage_sutter: MP-285 stage connection closed
2023-12-12 19:10:53,168 - model - INFO - camera_hamamatsu: HamamatsuOrca Shutdown
2023-12-12 19:10:53,168 - model - DEBUG - zoom_synthetic: SyntheticZoom Servo instance Deleted
2023-12-12 19:10:53,168 - model - DEBUG - remote_focus_equipment_solutions: Closing RemoteFocusEquipmentSolutions Serial Port
2023-12-12 19:10:53,188 - model - ERROR - stage_sutter: MP-285 - Stage stop failed: ClearCommError failed (OSError(9, 'The handle is invalid.', None, 6))
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\stages\stage_sutter.py", line 272, in stop
    self.stage.interrupt_move()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 383, in interrupt_move
    self.safe_write(bytes.fromhex("03"))
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 101, in safe_write
    self.serial.read_all()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialutil.py", line 652, in read_all
    return self.read(self.in_waiting)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (OSError(9, 'The handle is invalid.', None, 6))
2023-12-12 19:10:53,228 - model - DEBUG - stage_sutter: MP-285 stage connection closed
```
</details>

model.info
<details>

```
2023-12-12 19:10:35,602 - model - INFO - model: Navigate Model - Running data process, get frames [97]
2023-12-12 19:10:35,622 - model - INFO - model: Navigate Model - Sent through pipe97
2023-12-12 19:10:35,826 - model - INFO - daq_ni: Waveform Template Name: Default
2023-12-12 19:10:35,831 - model - INFO - daq_ni: Waveform Expand Num = 1
2023-12-12 19:10:35,831 - model - INFO - daq_ni: Waveform Repeat Num = 1
2023-12-12 19:10:36,081 - model - INFO - model: Navigate Model - Running data process, get frames [98]
2023-12-12 19:10:36,102 - model - INFO - model: Navigate Model - Sent through pipe98
2023-12-12 19:10:36,316 - model - INFO - daq_ni: Waveform Template Name: Default
2023-12-12 19:10:36,320 - model - INFO - daq_ni: Waveform Expand Num = 1
2023-12-12 19:10:36,320 - model - INFO - daq_ni: Waveform Repeat Num = 1
2023-12-12 19:10:36,578 - model - INFO - model: Navigate Model - Running data process, get frames [99]
2023-12-12 19:10:36,617 - model - INFO - model: Navigate Model - Sent through pipe99
2023-12-12 19:10:37,118 - model - INFO - model: Navigate Model - Running data process, get frames []
[... 31 lines cut ...]
2023-12-12 19:10:37,118 - controller - INFO - controller: Navigate Controller - Captured 3000, z-stack Images
2023-12-12 19:10:37,138 - controller - INFO - gui_controller: ChannelsTabController : ('acquisition mode has been changed to', 'stop')
[... cut ...]
2023-12-12 19:10:37,118 - controller - INFO - controller: Navigate Controller - Received Image: stop
2023-12-12 19:10:37,118 - controller - INFO - controller: Navigate Controller - Captured 3000, z-stack Images
2023-12-12 19:10:37,138 - controller - INFO - gui_controller: ChannelsTabController : ('acquisition mode has been changed to', 'stop')
2023-12-12 19:10:50,018 - controller - INFO - gui_controller: AcquireBarController : ('Exiting Program',)

```
</details>

Have I mentioned that I want to set all of the MP285 stages on fire?

annie-xd-wang, 2024-01-12:
The error doesn't show up on my computer with synthetic devices, and the waveforms are displayed correctly. It seems that there aren't any valid remote-focus waveforms. Does this error happen on other machines?

zacsimile, 2024-01-12:
Maybe the stage error disrupts the acquisition and then the waveforms cannot generate?

Surprised the stages don't work. Should be fine. Is Coherent Connection open? That tends to mess with it. Otherwise, power cycling and pressing the MOVE button on the stage controller should reset everything and it should work.

AdvancedImagingUTSW, 2024-01-12:
This works on the most recent develop branch on Voodoo.  Maybe it is CT-ASLM-V2 specific...

[image]

closed 2024-02-02 by AdvancedImagingUTSW