TheDeanLab/navigate #781: Validated Entry Error

AdvancedImagingUTSW, 2024-01-04:
**Describe the bug**
I'm on a roll. I directly changed the value for the end position in the stack acquisition settings (e.g., pressed delete before typing).

[image]

```
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\channels_tab_controller.py", line 343, in update_z_steps
    end_position = float(self.stack_acq_vals["end_position"].get())
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 568, in get
    return self._tk.getdouble(self._tk.globalgetvar(self._name))
_tkinter.TclError: expected floating-point number but got ""

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\tiling_wizard_controller2.py", line 135, in <lambda>
    "write", lambda *args: self.update_fov("z")
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\tiling_wizard_controller2.py", line 531, in update_fov
    z = float(self.stack_acq_widgets["end_position"].get()) - float(
ValueError: could not convert string to float: ''
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\channels_tab_controller.py", line 343, in update_z_steps
    end_position = float(self.stack_acq_vals["end_position"].get())
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 568, in get
    return self._tk.getdouble(self._tk.globalgetvar(self._name))
_tkinter.TclError: expected floating-point number but got ""

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\tiling_wizard_controller2.py", line 138, in <lambda>
    "write", lambda *args: self.update_fov("z")
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\tiling_wizard_controller2.py", line 531, in update_fov
    z = float(self.stack_acq_widgets["end_position"].get()) - float(
ValueError: could not convert string to float: ''
```

AdvancedImagingUTSW, 2024-01-05:
Only use validated widgets from here on. 

I also think a subset of the validated widgets have broken functionality when it comes to disable/readonly configuration states.

zacsimile, 2024-01-07:
I don't get this error on my Mac. I guess it is a difference in the trace binding order of `update_fov()` and `update_z_steps()` between Mac and Windows. `update_z_steps()` should already handle this error (see the `try`/`except` block starting on line 340. I've added a similar(ish) catch to `update_fov()` to address this error. Long term we can convert the `Spinbox`es to `ValidatedSpinbox`es.

closed 2024-01-07 by AdvancedImagingUTSW