TheDeanLab/navigate #784: Loading Multipositions

AdvancedImagingUTSW, 2024-01-05:
```
(base) C:\Users\Spectral>conda activate navigate

(navigate) C:\Users\Spectral>navigate
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\multi_position_controller.py", line 220, in load_positions
    self.table.redraw()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 488, in redraw
    self.redrawVisible(event, callback)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 470, in redrawVisible
    self.colorRows()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 581, in colorRows
    colors = rc[colname].loc[idx]
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 931, in __getitem__
    return self._getitem_axis(maybe_callable, axis=axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1153, in _getitem_axis
    return self._getitem_iterable(key, axis=axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1093, in _getitem_iterable
    keyarr, indexer = self._get_listlike_indexer(key, axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1314, in _get_listlike_indexer
    self._validate_read_indexer(keyarr, indexer, axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1374, in _validate_read_indexer
    raise KeyError(f"None of [{key}] are in the [{axis_name}]")
KeyError: "None of [Int64Index([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,\n            17, 18, 19, 20, 21, 22, 23],\n           dtype='int64')] are in the [index]"
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 320, in resized
    self.redrawVisible()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 470, in redrawVisible
    self.colorRows()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandastable\core.py", line 581, in colorRows
    colors = rc[colname].loc[idx]
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 931, in __getitem__
    return self._getitem_axis(maybe_callable, axis=axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1153, in _getitem_axis
    return self._getitem_iterable(key, axis=axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1093, in _getitem_iterable
    keyarr, indexer = self._get_listlike_indexer(key, axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1314, in _get_listlike_indexer
    self._validate_read_indexer(keyarr, indexer, axis)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\pandas\core\indexing.py", line 1374, in _validate_read_indexer
    raise KeyError(f"None of [{key}] are in the [{axis_name}]")
KeyError: "None of [Int64Index([ 0,  1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16,\n            17, 18, 19, 20, 21, 22, 23, 24],\n           dtype='int64')] are in the [index]"
```
Nothing useful found in the logs.

zacsimile, 2024-01-06:
Can you provide the multiposition CSV file you tried to load?

AdvancedImagingUTSW, 2024-01-06:
[multipositions.csv](https://github.com/TheDeanLab/navigate/files/13849793/multipositions.csv)

zacsimile, 2024-01-07:
Not getting this error on my Mac, but I am getting long wait times for the GUI to return control to the user after loading the table. I believe mine issues is the same issue as #450.

Did you try loading this `.csv` on other Windows machines?

closed 2024-01-10 by AdvancedImagingUTSW