TheDeanLab/navigate #785: load_dynamic_parameter_functions error

AdvancedImagingUTSW, 2024-01-05:
On the second time running this function, it failed...

<details>
<summary>Terminal Output</summary>
<br>

```
ge Max Intensity: 463 Percent Positive: 0.0
Image Max Intensity: 3786 Percent Positive: 0.00031640625
Image Max Intensity: 497 Percent Positive: 0.0
Image Max Intensity: 486 Percent Positive: 0.0
Image Max Intensity: 454 Percent Positive: 0.0
Image Max Intensity: 424 Percent Positive: 0.0
Image Max Intensity: 508 Percent Positive: 3.90625e-07
Image Max Intensity: 516 Percent Positive: 3.90625e-07
Image Max Intensity: 467 Percent Positive: 0.0
Image Max Intensity: 493 Percent Positive: 0.0
Exception inside ObjectInSubprocess: Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 526, in _child_loop
    result = getattr(obj, method_name)(*args, **kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 689, in run_command
    load_dynamic_parameter_functions(
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\feature_related_functions.py", line 392, in load_dynamic_parameter_functions
    load_dynamic_parameter_functions(item, feature_parameter_setting_path)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\feature_related_functions.py", line 379, in load_dynamic_parameter_functions
    ref_lib = parameter_config[parameter][args[idx]]
KeyError: <function hard_threshold at 0x0000020998DBCCA0>

model thread exception happened! <function hard_threshold at 0x0000020998DBCCA0> Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 251, in func
    target(*args, **kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 674, in <lambda>
    "model", lambda: self.model.run_command("load_feature", *args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 417, in attr
    return _get_response(self, True)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception: <function hard_threshold at 0x0000020998DBCCA0>

model thread ended because of exception!: not all arguments converted during string formatting Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 251, in func
    target(*args, **kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 674, in <lambda>
    "model", lambda: self.model.run_command("load_feature", *args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 417, in attr
    return _get_response(self, True)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception: <function hard_threshold at 0x0000020998DBCCA0>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 97, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 259, in func
    logger.debug(
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\log_files\filters.py", line 95, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

Exception in thread model:
Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 251, in func
    target(*args, **kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 674, in <lambda>
    "model", lambda: self.model.run_command("load_feature", *args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 417, in attr
    return _get_response(self, True)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception: <function hard_threshold at 0x0000020998DBCCA0>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 97, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 259, in func
    logger.debug(
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\log_files\filters.py", line 95, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 103, in run
    logger.debug(
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1434, in debug
    self._log(DEBUG, msg, args, **kwargs)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1589, in _log
    self.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1599, in handle
    self.callHandlers(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 1661, in callHandlers
    hdlr.handle(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 948, in handle
    rv = self.filter(record)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 806, in filter
    result = f.filter(record)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\log_files\filters.py", line 95, in filter
    if record.getMessage().startswith("Performance"):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\logging\__init__.py", line 367, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting

```
</details>

zacsimile, 2024-01-06:
Can you provide the logs?

AdvancedImagingUTSW, 2024-01-06:
I don't think the logs provided any insight. To replicate the error, we loaded our own function into the software, ran a customized workflow, stopped the acquisition part-way, and then tried to relaunch it. I'll replicate it once more and double-check the logs.

AdvancedImagingUTSW, 2024-01-07:
Yeah, nothing in the logs...

model_debug
```
2024-01-06 20:28:37,774 - model - INFO - model: Navigate Model - Running data process, get frames []
2024-01-06 20:28:37,774 - model - INFO - model: Navigate Model - Waiting 20
2024-01-06 20:28:37,774 - model - INFO - model: Navigate Model - Data thread stopped.
2024-01-06 20:28:37,774 - model - INFO - model: Navigate Model - Received frames in total: 433
2024-01-06 20:28:37,824 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -10671.6, 'x_pos': 13811.36, 'f_pos': 704.64}
2024-01-06 20:28:37,920 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -10671.6, 'x_pos': 13811.36, 'f_pos': 704.64}
2024-01-06 20:28:37,968 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -10671.6, 'x_pos': 13811.36, 'f_pos': 704.64}
```

view_controller_debug
```
2024-01-06 20:28:36,196 - controller - INFO - controller: Navigate Controller - Received Image: 28
2024-01-06 20:28:36,465 - controller - INFO - controller: Navigate Controller - Received Image: 29
2024-01-06 20:28:36,734 - controller - INFO - controller: Navigate Controller - Received Image: 30
2024-01-06 20:28:37,004 - controller - INFO - controller: Navigate Controller - Received Image: 31
2024-01-06 20:28:37,273 - controller - INFO - controller: Navigate Controller - Received Image: 32
2024-01-06 20:28:37,986 - controller - INFO - controller: Navigate Controller - command passed from child, load_feature, (0,)
2024-01-06 20:28:37,986 - controller - INFO - controller: Navigate Controller - command passed from child, stop_acquire, ()
2024-01-06 20:28:37,988 - controller - INFO - controller: Navigate Controller - Captured 433, live Images
2024-01-06 20:28:38,012 - controller - INFO - gui_controller: StageController : ('Set stage position',)
2024-01-06 20:28:38,017 - controller - INFO - gui_controller: ChannelsTabController : ('acquisition mode has been changed to', 'stop')
2024-01-06 20:29:37,396 - controller - WARNING - tiling_wizard_controller2: Controller - Tiling Wizard - could not convert string to float: ''
2024-01-06 20:29:57,237 - controller - INFO - gui_controller: MultiPositionController : ('exporting csv file', 'E:/Kevin/multipositions2.csv')
2024-01-06 20:30:04,364 - controller - INFO - controller: Navigate Controller - command passed from child, load_feature, (9,)
```

I think it is this:
```
2024-01-06 20:30:04,364 - controller - INFO - controller: Navigate Controller - command passed from child, load_feature, (9,)
```

annie-xd-wang, 2024-01-09:
I loaded two testing functions and tried several times to use them as parameters. There were no errors. However, if I typed an unloaded function name, it did pop up an error similar to what is shown here. Need to double-confirm how to reproduce the error.

closed 2024-01-17 by AdvancedImagingUTSW