TheDeanLab/navigate #793: 'DetectTissueInStackAndRecord' object has no attribute 'detect_tissue_queue'

AdvancedImagingUTSW, 2024-01-10:
Because I was struggling with the feature where we eliminate tiles on the fly, I tried the `Remove Empty Tiles` feature. It through this error:

```
2024-01-10 14:18:17,845 - model - DEBUG - image_writer: MIP Directory Created - Z:\bioinformatics\Danuser_lab\Dean\dean\2024-01-08\Kevin\BoneMarrow\LepR\DsRed\2024-01-10\Cell_006\MIP
2024-01-10 14:18:18,068 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:18:18,075 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Moving to Position 0
2024-01-10 14:18:18,143 - model - INFO - daq_ni: Waveform Template Name: Default
2024-01-10 14:18:18,148 - model - INFO - daq_ni: Waveform Expand Num = 1
2024-01-10 14:18:18,149 - model - INFO - daq_ni: Waveform Repeat Num = 1
2024-01-10 14:18:18,208 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12256.880000000001, 'x_pos': 8757.16, 'f_pos': -2045.64}
2024-01-10 14:18:18,417 - model - INFO - model: Navigate Model - Running data process, get frames [0]
2024-01-10 14:18:18,417 - model - INFO - model: Navigate Model - Sent through pipe0
2024-01-10 14:18:18,470 - model - DEBUG - feature_container: running signal node: MoveToNextPositionInMultiPositionTable
2024-01-10 14:18:18,470 - model - DEBUG - feature_container: running signal node: MoveToNextPositionInMultiPositionTable
2024-01-10 14:18:18,470 - model - DEBUG - common_features: multi-position current idx: 0, 240
2024-01-10 14:18:18,473 - model - DEBUG - common_features: MoveToNextPositionInMultiPosition: {'x': 8757.16, 'y': -12483.6, 'z': -100.0, 'theta': 0.0, 'f': -2585.76}
2024-01-10 14:18:18,732 - model - DEBUG - common_features: MoveToNextPositionInMultiPosition: move done
2024-01-10 14:18:18,752 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:18,918 - model - DEBUG - HamamatsuAPI: error message: -2147483386
2024-01-10 14:18:18,918 - model - INFO - model: Navigate Model - Running data process, get frames []
2024-01-10 14:18:18,919 - model - INFO - model: Navigate Model - Waiting 20
2024-01-10 14:18:18,958 - model - INFO - model: Navigate Model - Running data process, get frames [1]
2024-01-10 14:18:18,959 - model - INFO - model: Navigate Model - Sent through pipe1
2024-01-10 14:18:19,012 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,016 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,016 - model - DEBUG - remove_empty_tiles: move to position (z, f): (-200.0, -2585.64), 0, 2
2024-01-10 14:18:19,152 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:19,357 - model - INFO - model: Navigate Model - Running data process, get frames [2]
2024-01-10 14:18:19,390 - model - DEBUG - remove_empty_tiles: *** this frame has enough percentage of tissue!2
2024-01-10 14:18:19,390 - model - INFO - model: Navigate Model - Sent through pipe2
2024-01-10 14:18:19,411 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,411 - model - DEBUG - remove_empty_tiles: *** detect tissue signal end function: 1
2024-01-10 14:18:19,411 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,411 - model - DEBUG - remove_empty_tiles: move to position (z, f): (-150.0, -2585.64), 1, 3
2024-01-10 14:18:19,552 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:19,758 - model - INFO - model: Navigate Model - Running data process, get frames [3]
2024-01-10 14:18:19,758 - model - INFO - model: Navigate Model - Sent through pipe3
2024-01-10 14:18:19,811 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,811 - model - DEBUG - remove_empty_tiles: *** detect tissue signal end function: 2
2024-01-10 14:18:19,811 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:19,811 - model - DEBUG - remove_empty_tiles: move to position (z, f): (-100.0, -2585.64), 2, 4
2024-01-10 14:18:19,952 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:20,159 - model - INFO - model: Navigate Model - Running data process, get frames [4]
2024-01-10 14:18:20,159 - model - INFO - model: Navigate Model - Sent through pipe4
2024-01-10 14:18:20,212 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:20,212 - model - DEBUG - remove_empty_tiles: *** detect tissue signal end function: 3
2024-01-10 14:18:20,212 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:20,212 - model - DEBUG - remove_empty_tiles: move to position (z, f): (-50.0, -2585.64), 3, 5
2024-01-10 14:18:20,352 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:20,558 - model - INFO - model: Navigate Model - Running data process, get frames [5]
2024-01-10 14:18:20,558 - model - INFO - model: Navigate Model - Sent through pipe5
2024-01-10 14:18:20,612 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:20,612 - model - DEBUG - remove_empty_tiles: *** detect tissue signal end function: 4
2024-01-10 14:18:20,612 - model - DEBUG - feature_container: running signal node: DetectTissueInStackAndRecord
2024-01-10 14:18:20,612 - model - DEBUG - remove_empty_tiles: move to position (z, f): (0.0, -2585.64), 4, 6
2024-01-10 14:18:20,752 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -2585.64}
2024-01-10 14:18:20,958 - model - INFO - model: Navigate Model - Running data process, get frames [6]
2024-01-10 14:18:20,983 - model - DEBUG - feature_container: DataContainer - Traceback (most recent call last):
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\feature_container.py", line 644, in run
    result, is_end = self.curr_node.run(*args)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\feature_container.py", line 377, in run
    if self.node_type == "multi-step" and not self.node_funcs["end"]():
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\remove_empty_tiles.py", line 481, in end_func_data
    return super().end_func_data()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\features\remove_empty_tiles.py", line 330, in end_func_data
    self.detect_tissue_queue.put(self.has_tissue_flag)
AttributeError: 'DetectTissueInStackAndRecord' object has no attribute 'detect_tissue_queue'

```

annie-xd-wang, 2024-01-10:
On which branch were you working when this error appeared?

AdvancedImagingUTSW, 2024-01-10:
I am on `Develop`, albeit 8 commits behind.

`model_debug.log` is here:
<details>
```
2024-01-10 14:16:50,114 - model - INFO - camera_hamamatsu: HamamatsuOrca Initialized
2024-01-10 14:16:50,166 - model - DEBUG - remote_focus_equipment_solutions: RemoteFocusEquipmentSolutions - Opening Voice Coil on COM: COM7
2024-01-10 14:16:50,166 - model - DEBUG - remote_focus_equipment_solutions: RemoteFocusEquipmentSolutions - Sending command: b'd0\r'
2024-01-10 14:16:51,417 - model - DEBUG - remote_focus_equipment_solutions: RemoteFocusEquipmentSolutions - Received command: b'>!214\n'
2024-01-10 14:16:56,418 - model - DEBUG - remote_focus_equipment_solutions: RemoteFocusEquipmentSolutions - Servo Engaged
2024-01-10 14:16:56,517 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Placing device In Online Mode
2024-01-10 14:16:56,537 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Initialized.
2024-01-10 14:16:56,537 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Moving to Position 0
2024-01-10 14:16:56,588 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel -  Placed in Default Filter Position.
2024-01-10 14:16:56,621 - model - DEBUG - zoom_synthetic: SyntheticZoom Servo Initialized
2024-01-10 14:16:57,023 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:16:57,208 - model - INFO - camera_hamamatsu: HamamatsuOrca - subarray_hpos, 324.0
2024-01-10 14:16:57,209 - model - INFO - camera_hamamatsu: HamamatsuOrca - subarray_hsize, 1400.0
2024-01-10 14:16:57,209 - model - INFO - camera_hamamatsu: HamamatsuOrca - subarray_vpos, 224.0
2024-01-10 14:16:57,209 - model - INFO - camera_hamamatsu: HamamatsuOrca - subarray_vsize, 1600.0
2024-01-10 14:16:57,825 - model - DEBUG - zoom_base: Changed Zoom to 36X
2024-01-10 14:16:57,825 - model - DEBUG - zoom_base: Zoom position: None
2024-01-10 14:16:57,825 - model - DEBUG - zoom_base: Zoom position: None
2024-01-10 14:16:57,825 - model - DEBUG - model: Change zoom of CTASLMv2 to 36X
2024-01-10 14:16:57,871 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:17:09,555 - model - DEBUG - zoom_base: Changed Zoom to 36X
2024-01-10 14:17:09,555 - model - DEBUG - zoom_base: Zoom position: None
2024-01-10 14:17:09,555 - model - DEBUG - zoom_base: Zoom position: None
2024-01-10 14:17:09,555 - model - DEBUG - model: Change zoom of CTASLMv2 to 36X
2024-01-10 14:17:09,601 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:17:09,649 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:17:16,227 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:17:16,744 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:16,751 - model - DEBUG - filter_wheel_sutter: SutterFilterWheel - Moving to Position 0
2024-01-10 14:17:16,878 - model - INFO - daq_ni: Waveform Template Name: Default
2024-01-10 14:17:16,884 - model - INFO - daq_ni: Waveform Expand Num = 1
2024-01-10 14:17:16,884 - model - INFO - daq_ni: Waveform Repeat Num = 1
2024-01-10 14:17:16,911 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-10 14:17:16,930 - model - DEBUG - stage_sutter: MP-285 - Position: {'y_pos': -12483.48, 'x_pos': 8757.16, 'f_pos': -0.04}
2024-01-10 14:17:17,142 - model - INFO - model: Navigate Model - Running data process, get frames [0]
2024-01-10 14:17:17,142 - model - INFO - model: Navigate Model - Sent through pipe0
2024-01-10 14:17:17,193 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:17,193 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-10 14:17:17,403 - model - INFO - model: Navigate Model - Running data process, get frames [1]
2024-01-10 14:17:17,403 - model - INFO - model: Navigate Model - Sent through pipe1
2024-01-10 14:17:17,454 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:17,454 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-10 14:17:17,664 - model - INFO - model: Navigate Model - Running data process, get frames [2]
2024-01-10 14:17:17,664 - model - INFO - model: Navigate Model - Sent through pipe2
2024-01-10 14:17:17,716 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:17,716 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-10 14:17:17,926 - model - INFO - model: Navigate Model - Running data process, get frames [3]
2024-01-10 14:17:17,926 - model - INFO - model: Navigate Model - Sent through pipe3
2024-01-10 14:17:17,977 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:17,977 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-01-10 14:17:18,186 - model - INFO - model: Navigate Model - Running data process, get frames [4]
2024-01-10 14:17:18,186 - model - INFO - model: Navigate Model - Sent through pipe4
2024-01-10 14:17:18,238 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-01-10 14:17:18,238 - model - DEBUG - feature_container: running signal node: LoopByCount
[... cut ...]
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialutil.py", line 652, in read_all
    return self.read(self.in_waiting)
  File "C:\Users\Spectral\.conda\envs\navigate\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (OSError(9, 'The handle is invalid.', None, 6))
2024-01-10 14:25:25,003 - model - DEBUG - stage_sutter: MP-285 stage connection closed

</details>

annie-xd-wang, 2024-01-10:
The file was somehow messed up, but it's fixed now.

closed 2024-01-10 by AdvancedImagingUTSW