TheDeanLab/navigate #798: Experiment file not entirely accurate

AdvancedImagingUTSW, 2024-01-15:
While looking at two experiment files from Kartik's lab, I realized that the devices section is not accurate. For example, all imaging was done on the BT-MesoSPIM, which has an ASI filter wheel and stage, and does not have a zoom.

```
    "Devices": {
        "daq": "NI",
        "camera": "HamamatsuOrca",
        "etl": "ETL",
        "filter_wheel": "Sutter",
        "stage": "PI",
        "zoom": "Dynamixel"
    },
```

AdvancedImagingUTSW, 2024-01-15:
Also, it would be nice if the galvo settings were at least provided here too. I was looking to see if they did different scan ranges between experiments, and this information isn't available to me.

annie-xd-wang, 2024-01-17:
The 'Devices' section is no longer used and it can be removed from the 'experiment.yml' file. Have you checked the "waveform_constants.yml" file? There are some galvo settings. We can save a copy of 'waveform_constants.yml' the same way as 'experiment.yml' if it has the information you need.

AdvancedImagingUTSW, 2024-01-19:
So, I need to delete the devices section from the repository, including local versions on our machines.

For the `waveform_constants.yml` file, it could make sense to also save this with the data. But, would it make sense to also provide a way to load this file, if we are saving it? Or is just for references?

closed 2024-01-24 by AdvancedImagingUTSW