TheDeanLab/navigate #817: Timing observations

AdvancedImagingUTSW, 2024-02-05:
From @bojuichang

Dear Kevin and Zach,

First, thank you for fixing the little bug in the Navigate. I will try to submit the bugs and the requests on GitHub in the future, but for the sake of time, let me just report what I have observed recently.
 
So, at the beginning, I found that there's a delay in the acquisition. Simply speaking, if I set a 50ms exposure time, the real acquisition time is not really 50ms, but rather close to 66ms. This is observed by checking either the trigger signal or the laser modulation signal with an oscilloscope. Kevin fixed that, but every time when Kevin left, I observed that delay again.  
 
Zach and I have had a few discussions through Teams this week. He suggested me to check the timings in normal vs light-sheet camera mode. Interestingly, indeed, the delay is much less in the light-sheet mode. I guess this is why every time after Kevin left, I observed that delay again because I always start with the normal camera mode.
 
I heard this might be an inherent design flaw. If it is really too difficult to fix, I think maybe it is fine. Because if someone uses Voodoo, they probably will use the light-sheet (ASLM) mode, not the normal (Stelzer) mode.
 
However, I did some test today and I used the oscilloscope to measure the timing of the trigger signal, as attached. You will see even in the light-sheet mode, you can still observe ~7ms and ~10ms delay with 50ms and 100ms exposure time, respectively, in the continuous acquisition. And in the Z stack acquisition mode, the delay got longer. @Kevin, I think we did not see that delay because 7ms or 10ms delay is hard to tell by eyes. But with the measurement from the oscilloscope, it's obvious.
 
The other thing I observed was that in the X-galvo signal, the percent delay does not affect the total delay, but the percent smoothing or settle duration would increase the duration. This is a little odd to me because I thought the percent delay should have an effect, but not the percent smoothing. But this is okay, I set the percent smoothing and the settle duration to zero now.
 
I hope this helps.

Best,
Bo-Jui

AdvancedImagingUTSW, 2024-02-05:
Hi Bo-Jui,
 
Thanks for the feedback!
 
For the normal vs. light-sheet issue, I don't think this part is a design flaw. However, I am not sure. I am curious to hear your take on our reasoning for introducing this delay in the code. We wrote the code this way because of how Hamamatsu's edge trigger mode works (see figure below from https://www.hamamatsu.com/content/dam/hamamatsu-photonics/sites/static/sys/en/manual/C13440-20CU_IM_En.pdf).
 
[image] 

The way I interpret this graph, some of the lines of the camera start their exposure time later than the initial trigger, and so it takes longer than the integration time for the camera to finish acquiring. We found that this seemed to be true experimentally, but perhaps I interpreted the results wrong. What I found was that if we did not add a delay to the normal mode, and we stopped both the camera and the galvo at the same time, an image of a parked beam would superimpose itself on the final image. I interpreted this to mean the camera was still acquiring after we'd stopped it, even though it had been at least a full exposure time since the acquisition started. When we added the delay and kept the galvo going through the delay, the image acquired looked normal. As such, I think we do need this delay in normal mode. I am not sure how Gerard wrote up his normal acquisition, and would be curious to know.
 
For the 7/10 ms delay, could you try 200 ms integration time as well? Or did you try measuring the delay three times at a single exposure time? I'd guess these are all somewhere around 7-12 ms for any exposure time. I suspect this is a variable delay time introduced by either 1) writing to the DAQ or 2) querying the configuration dictionary, but it could be any of the other calls in the image acquisition process. We should chase down exactly what is causing this. If the Z-stack is taking longer, this may be due either to additional configuration dictionary calls, multiprocessing events/locks, or something else entirely. (I've been listing first the things we know to have caused long delays in the past, I am not sure they still do.)
 
This is where the design flaw might be, in that we mix hardware and software timing. We may need to adopt a single write at the beginning of the acquisition (as we do in Confocal Projection mode) for the Voodoo in order to image at a reasonable rate.
 
For the galvo...
 
1) Looking at the galvo_ni.py code, the percent delay in the GUI is not used by the galvo, instead the galvo uses the camera delay percent. I cannot remember why we do this and I'm not sure it's correct. 
 
2) For the percent smoothing, I'd argue it should extend the waveform a bit. Smoothing is implemented as a convolution operation between the original signal (of length M) and the smoothing signal (square function of length N). A convolution of two functions of lengths M and N is of length M + N - 1. It also makes sense intuitively that we'd need more samples to make a sharp turn in a function (say around the peak of a triangle wave) more smoothly. If we clip back to length M, we'd lose some of our desired signal. In this case, it would effectively change the sampling rate of the galvo sweeping across the frame.
 
Did Gerard implement the smoothing differently? Do you have a way to smooth the function without increasing the sample length (I have one, but it comes at the cost of changing where the linear portion of the function is, which is also undesirable)?
 
Thanks, again, and I look forward to hearing your thoughts on these responses. 
 
Best,
Zach

AdvancedImagingUTSW, 2024-02-05:
Just to reply to Zach's question. 
For the measurement, I try both continuous mode and Z stack mode. For the Z stack, I just let it repeat 20 times without any delay, so I have time to look at what the oscilloscope measures.
For each measurement, yes, you can see a variation of +- 1ms. So, when I measure 66ms, it's most likely 66+- 1ms.  
I updated the txt file as attached. As you can see there, when applying 200ms exposure time, the delay got longer. For example, in the continuous scan with light-sheet mode, the measured period is  57ms, 109ms, and 215ms with exposure time of 50ms, 100ms, and 200ms, respectively.

I also found something interesting. In the normal mode. The timing is affected by the size of ROI (e.g. 2048*2048 vs 1024*1024). However, in light-sheet mode, the timing is not affected by the size of the ROI. 

And then, I also wonder why the waveform (i.e. the percent smoothing or the settle duration) of the X-galvo affects the timing in the normal mode since that has been disabled. Well, the settle duration might affect the timing depending on how you write your code, but percent smoothing? But this is not the main issue now. I can live with it by just setting the percent smoothing and the settle duration to zero.

I don't know exactly how Gerard wrote his code. I might also have my own LabVIEW code to smooth the waveform. I will have a look and let you know if I have some thoughts.

[Latency investigate_20240126.txt](https://github.com/TheDeanLab/navigate/files/14159132/Latency.investigate_20240126.txt)

annie-xd-wang, 2024-02-05:
In the normal mode, there is a camera readout time added to each frame, which is affected by the image size. 
A camera delay, defined as a percentage parameter and configured in the configuration.yaml file, is applied to the entire sweeping time duration. Additionally, a duration for the remote focus to transition back is added to the entire sweeping time. It's also a percentage parameter. This setup is the primary reason for extended delays as the exposure time increases.

AdvancedImagingUTSW, 2024-03-01:
Confirmed to be working on the following cameras:

- [ ] Orca Lightning
- [ ] Orca Fusion
- [ ] Orca Flash 4.0
- [ ] Orca Fire

AdvancedImagingUTSW, 2024-03-01:
Documentation here:

https://www.hamamatsu.com/jp/en/product/cameras/cmos-cameras.html

AdvancedImagingUTSW, 2024-03-01:
[image]

[image]

BT and regular fusion have same timing in light-sheet mode.

bojuichang, 2024-03-01:
> EXTERNAL MAIL
>
> image.png (view on web)<https://urldefense.com/v3/__https://github.com/TheDeanLab/navigate/assets/42547789/ad36875e-9674-4ecc-b15b-8b7cd9ccb0dd__;!!MznTZTSvDXGV0Co!EpjV2Uby9AW7mFX6DDtB-sWE4vRCcNsj9WVYW37hWvP5TPOI6eE6MMYvlZ9DhoZRiljiq-W1YAWzdPR-EyBOOc_sqYUMtyzQSaF6Cg$>
>
> image.png (view on web)<https://urldefense.com/v3/__https://github.com/TheDeanLab/navigate/assets/42547789/ed06a086-67ef-42e1-80be-6a43ca92a401__;!!MznTZTSvDXGV0Co!EpjV2Uby9AW7mFX6DDtB-sWE4vRCcNsj9WVYW37hWvP5TPOI6eE6MMYvlZ9DhoZRiljiq-W1YAWzdPR-EyBOOc_sqYUMtyzHD-9J-Q$>

[like]  Bo-Jui Chang reacted to your message:

CAUTION: This email originated from outside UTSW. Please be cautious of links or attachments, and validate the sender's email address before replying.

UT Southwestern

Medical Center

The future of medicine, today.

AdvancedImagingUTSW, 2024-03-12:
Addressed in #808

closed 2024-03-12 by AdvancedImagingUTSW