TheDeanLab/navigate #827: Enhance Multi-Camera Operation

AdvancedImagingUTSW, 2024-02-09:
While we do support the simultaneous operation of multiple cameras, we could implement several features that would make it more useful. 

## Improved Display
On our original LabView software, the user could pop out a frame for each camera. This included the necessary controls for adjusting the intensity, lookup table, etc. This would be nice to implement.

That leaves the original image display, which remains within the main GUI by default. For multiple cameras, it would be nice to have the ability to create an overlay between each of the images, each in a user specified lookup table to distinguish each challenge. Perhaps the properties of the display could be pulled from the other windows (e.g., max counts, autoscale, etc.). 

## Data Saving
Currently we create multiple instances of the Image Writers to save the data. This should be upgraded so that we can save multiple channels with proper BioFormats Metadata. 

Any other suggestions are welcome @bojuichang

AdvancedImagingUTSW, 2024-03-08:
A few bits of the documentation from our old LabView code for inspiration: 
[Uploading LouisXIV Microscope - Software User Manual.pdf…]()

AdvancedImagingUTSW, 2024-03-08:
My current thoughts...

In the case of 1 camera operation, we don't have to do anything. Everything can be loaded as is.

In the case of more than 1 camera, I would like to create a tab for the overlay image, and one tab for each individual camera. The tabs should be able to pop out, just as we already do. 

Each tab, for each camera, should only refresh if it is popped out or if it is top-level. This will avoid unnecessary calculations as our image viewer is a bit resource intensive.

Ideally, we would be able to use abstraction to our advantage for the overlay camera. We could create a new class, that handles overlay images, which inherits most of the functions from the regular `CameraViewController`. 

As you can see in the documentation above, there are a few distinct widgets that are helpful in the overview mode. For example, there are sliders that allow you to adjust the contrast on a per-channel basis.

bojuichang, 2024-03-19:
Sorry for being late. So is there now a version for two camera operations?
The features of our original LabVIEW code work fine to me.

AdvancedImagingUTSW, 2024-03-19:
@annie-xd-wang Could you provide some basic instructions on how to get multiple cameras operating as our code is? I'll format it and make a page on our documentation from it.

AdvancedImagingUTSW, 2024-03-20:
@bojuichang - #849 provided documentation on how to enable and disable the multi camera operation mode. Please see the directions [here](https://thedeanlab.github.io/navigate/user_guide/setup_microscope.html) and let us know if anything is not clear.  Additional changes to the code are still needed in order to operate with an overlay, etc.