TheDeanLab/navigate #833: Tab of laser power control

bojuichang, 2024-02-23:
I always need to be careful whether I enter 10% or 100% of the laser power. Because the size of the tab to enter the laser power is simply just too small. 

Can you make it larger? 
Thanks.

[image]

bojuichang, 2024-02-28:
I feel I don't see the change. The tab is still small.
[image]

AdvancedImagingUTSW, 2024-03-01:
If I comment out the width value, it is supposed to default to a value of ~20 according to ttk documentation. This appears to be true:

```
            #  Laser Power Spinbox
            self.laserpower_variables.append(tk.StringVar())
            self.laserpower_pulldowns.append(
                ValidatedSpinbox(
                    self.frame_columns[2],
                    from_=0,
                    to=100.0,
                    textvariable=self.laserpower_variables[num],
                    increment=5,
                    #width=5,
                    font=tk.font.Font(size=11),
                )
            )
```
[image]

If I return the value to width to 5, I think it looks good.  It was previously set to 4, which might have been just a bit too tight.
[image]

bojuichang, 2024-03-01:
Thanks for the effort. 
Is it possible to have something in between, like 4.2 or 4.5? I feel the width of 5 is a bit too large. ^^! The width of 4 seems to be a bit tight but personally, I think it is better. Or, maybe you can have the width of 4 and make the font smaller, like the one in the Laser or Filter settings. Nevertheless, both work for me. 
Thanks a lot.

AdvancedImagingUTSW, 2024-03-01:
The value must be an integer, and changing the font size can be tricky since it can disrupt the spacing and alignment of the rows and columns. that's why the Laser and Filter settings have different font sizes; it aligns to the other variables in that section. I went ahead and merged it.

closed 2024-03-01 by AdvancedImagingUTSW