TheDeanLab/navigate #839: BT_mesoSPIM_Galvo_0_Frequency_Faied_to_update

JinlongL, 2024-03-05:
In recent branch, while changing the galvo 0 frequency on BT- mesoSPIM, the software update value weirdly. As seen in picture, 74 Hz frequency is the same as 5 Hz frequency. While switched to older branch, the problem solved. An error message on camera shows: -2147483386.
[image]
[image]

```
2024-03-05 15:14:27,290 - model - INFO - model: Navigate Model - Stopping with stop command.
2024-03-05 15:14:27,455 - model - INFO - model: Navigate Model - Running data process, get frames [21]
2024-03-05 15:14:27,455 - model - INFO - model: Navigate Model - Sent through pipe21
2024-03-05 15:14:27,456 - model - INFO - model: Navigate Model - Data thread stopped.
2024-03-05 15:14:27,456 - model - INFO - model: Navigate Model - Received frames in total: 22
2024-03-05 15:14:28,861 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-03-05 15:14:28,894 - model - INFO - daq_ni: Waveform Template Name: Default
2024-03-05 15:14:28,900 - model - INFO - daq_ni: Waveform Expand Num = 1
2024-03-05 15:14:28,900 - model - INFO - daq_ni: Waveform Repeat Num = 1
2024-03-05 15:14:29,362 - model - DEBUG - HamamatsuAPI: error message: -2147483386
2024-03-05 15:14:29,362 - model - INFO - model: Navigate Model - Running data process, get frames []
2024-03-05 15:14:29,362 - model - INFO - model: Navigate Model - Waiting 20
2024-03-05 15:14:29,864 - model - DEBUG - HamamatsuAPI: error message: -2147483386
2024-03-05 15:14:29,864 - model - INFO - model: Navigate Model - Running data process, get frames []
2024-03-05 15:14:29,864 - model - INFO - model: Navigate Model - Waiting 19
2024-03-05 15:14:30,038 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-03-05 15:14:30,281 - model - INFO - model: Navigate Model - Running data process, get frames [0]
2024-03-05 15:14:30,281 - model - INFO - model: Navigate Model - Sent through pipe0
2024-03-05 15:14:30,305 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-03-05 15:14:30,305 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-03-05 15:14:30,514 - model - INFO - model: Navigate Model - Running data process, get frames [1]
2024-03-05 15:14:30,514 - model - INFO - model: Navigate Model - Sent through pipe1
2024-03-05 15:14:30,538 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-03-05 15:14:30,538 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-03-05 15:14:30,749 - model - INFO - model: Navigate Model - Running data process, get frames [2]
2024-03-05 15:14:30,749 - model - INFO - model: Navigate Model - Sent through pipe2
2024-03-05 15:14:30,772 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-03-05 15:14:30,772 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-03-05 15:14:30,953 - model - INFO - model: Navigate Model - Stopping with stop command.
2024-03-05 15:14:30,982 - model - INFO - model: Navigate Model - Running data process, get frames [3]
2024-03-05 15:14:30,982 - model - INFO - model: Navigate Model - Sent through pipe3
2024-03-05 15:14:30,982 - model - INFO - model: Navigate Model - Data thread stopped.
2024-03-05 15:14:30,982 - model - INFO - model: Navigate Model - Received frames in total: 4
```

AdvancedImagingUTSW, 2024-03-05:
In what recent branch?

JinlongL, 2024-03-05:
Both develop branch and the fixed-timing branch. I will try the fix_exposure_time_error branch later.

EXTERNAL MAIL

CAUTION: This email originated from outside UTSW. Please be cautious of links or attachments, and validate the sender's email address before replying.

UT Southwestern

Medical Center

The future of medicine, today.

JinlongL, 2024-03-05:
The fix_exposure_time_error branch Annie just made fixed the issue.

AdvancedImagingUTSW, 2024-03-06:
@annie-xd-wang - Will this be a separate PR, or part of the timing overhaul?

closed 2024-03-07 by AdvancedImagingUTSW