TheDeanLab/navigate #858: Error message when saving binned images

mehr0096, 2024-04-23:
When saving binned images (single acquisition), an error message shows up that there is insufficient disk space. Issue reproduced with Hamamatsu and Photometrics camera on two setups.

DaetwylerStephan, 2024-04-24:
Another question: is the function "get_minimum_waiting_time" in the camera classes needed?

AdvancedImagingUTSW, 2024-04-24:
@annie-xd-wang, can you confirm? I just looked and it appears that the function "get_minimum_waiting_time" is defined and tested, but not used. If that is correct, then we can remove the method and the tests that evaluate it.

DaetwylerStephan, 2024-04-24:
The Photometrics camera asks for an exposure time when setting up the initialize_image_series:
Function from Photometrics API: 
start_live(self, **exp_time**=None, buffer_frame_count=16, stream_to_disk_path=None):
        """Calls the pvc.start_live function to setup a circular buffer acquisition.

In the software, however, the exposure times are only calculated afterwards. I don't know how to best combine this. Any thoughts? @zacsimile @annie-xd-wang

annie-xd-wang, 2024-04-24:
@DaetwylerStephan and @AdvancedImagingUTSW, get_minimum_waiting_time() isn't used anymore and removed in the PR.

annie-xd-wang, 2024-04-24:
@DaetwylerStephan , navigate will call camera.set_exposure_time() before acquiring data for each channel. Is there an API support changing the exposure time? If so, you could set up the buffer with a None exposure value in the function initialize_image_series().

DaetwylerStephan, 2024-04-24:
I tried the None value as exposure time. It doesn't work then

DaetwylerStephan, 2024-04-24:
Where/when would you set the exposure time then? I can set up the buffer with None exposure value but if I then update the exposure time function in set_exposure_time, it also seems too late.

closed 2024-04-25 by AdvancedImagingUTSW