TheDeanLab/navigate #860: Changing ASI FIlter While Continuous Scan Acquiring

EvoleneP, 2024-04-24:
**Describe the bug**
Was acquiring data in the live mode and I tried to change the emission filter, and got the following error. Did not originally happen if the acquisition was stopped. Will confirm.

```
base) C:\Users\Nikon>conda activate navigate

(navigate) C:\Users\Nikon>navigate
Exception inside ObjectInSubprocess: Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 526, in _child_loop
    result = getattr(obj, method_name)(*args, **kwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\model.py", line 624, in run_command
    waveform_dict = self.active_microscope.calculate_all_waveform()
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\microscope.py", line 478, in calculate_all_waveform
    stage.switch_mode("normal", exposure_times, sweep_times)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\devices\stages\stage_galvo.py", line 298, in switch_mode
    self.move_axis_absolute(self.axes[0], float(self.configuration["experiment"]["StageParameters"][self.axes[0]]))
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\devices\stages\stage_galvo.py", line 239, in move_axis_absolute
    self.ao_task.write(volts, auto_start=True)
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\site-packages\nidaqmx\task.py", line 1233, in write
    return _write_analog_f_64(
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\site-packages\nidaqmx\_task_modules\write_functions.py", line 32, in _write_analog_f_64
    check_for_error(error_code)
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Requested Sample Timing Type is not allowed, because there is already another task with analog output channels from the same device configured for a different Sample Timing Type. This is not supported on this device.
Change your application so that all the channels from this device are used in one task, set Sample Timing Type to On Demand for all tasks, or consider using two devices for the two tasks.
Task Name: _unnamedTask<0>

Status Code: -200963

model thread exception happened! Requested Sample Timing Type is not allowed, because there is already another task with analog output channels from the same device configured for a different Sample Timing Type. This is not supported on this device.
Change your application so that all the channels from this device are used in one task, set Sample Timing Type to On Demand for all tasks, or consider using two devices for the two tasks.
Task Name: _unnamedTask<0>

Status Code: -200963 Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 252, in func
    target(*args, **kwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 679, in <lambda>
    "model", lambda: self.model.run_command("update_setting", *args)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 417, in attr
    return _get_response(self, True)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception: Requested Sample Timing Type is not allowed, because there is already another task with analog output channels from the same device configured for a different Sample Timing Type. This is not supported on this device.
Change your application so that all the channels from this device are used in one task, set Sample Timing Type to On Demand for all tasks, or consider using two devices for the two tasks.
Task Name: _unnamedTask<0>

Status Code: -200963

model thread ended because of exception!: not all arguments converted during string formatting Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 252, in func
    target(*args, **kwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 679, in <lambda>
    "model", lambda: self.model.run_command("update_setting", *args)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 417, in attr
    return _get_response(self, True)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception: Requested Sample Timing Type is not allowed, because there is already another task with analog output channels from the same device configured for a different Sample Timing Type. This is not supported on this device.
Change your application so that all the channels from this device are used in one task, set Sample Timing Type to On Demand for all tasks, or consider using two devices for the two tasks.
Task Name: _unnamedTask<0>

Status Code: -200963

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 98, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\thread_pool.py", line 260, in func
[... 109 lines cut ...]
024-04-24 09:56:42,761 - controller - INFO - controller: Navigate Controller - Received Image: 99
2024-04-24 09:56:42,861 - controller - INFO - controller: Navigate Controller - Received Image: 0
2024-04-24 09:56:42,965 - controller - INFO - controller: Navigate Controller - Received Image: 1
2024-04-24 09:56:43,028 - controller - INFO - gui_controller: ChannelSettingController : ('channel setting has been changed',)
2024-04-24 09:56:43,064 - controller - INFO - controller: Navigate Controller - Received Image: 2
2024-04-24 09:56:43,166 - controller - INFO - controller: Navigate Controller - Received Image: 3
2024-04-24 09:56:43,266 - controller - INFO - controller: Navigate Controller - Received Image: 4
2024-04-24 09:56:43,367 - controller - INFO - controller: Navigate Controller - Received Image: 5
2024-04-24 09:56:43,468 - controller - INFO - controller: Navigate Controller - Received Image: 6
2024-04-24 09:56:43,535 - controller - INFO - gui_controller: ChannelsTabController : ('Received command from child', 'update_setting', ('channel',))
2024-04-24 09:56:43,567 - controller - INFO - controller: Navigate Controller - Received Image: 7
2024-04-24 09:56:43,669 - controller - INFO - controller: Navigate Controller - Received Image: 8
2024-04-24 09:56:43,771 - controller - INFO - controller: Navigate Controller - Received Image: 9
2024-04-24 09:56:43,871 - controller - INFO - controller: Navigate Controller - Received Image: 10
2024-04-24 09:56:43,972 - controller - INFO - controller: Navigate Controller - Received Image: 11
2024-04-24 09:56:44,072 - controller - INFO - controller: Navigate Controller - Received Image: 12
2024-04-24 09:56:44,172 - controller - INFO - controller: Navigate Controller - Received Image: 13
2024-04-24 09:56:44,273 - controller - INFO - controller: Navigate Controller - Received Image: 14
2024-04-24 09:56:44,373 - controller - INFO - controller: Navigate Controller - Received Image: 15
2024-04-24 09:56:44,473 - controller - INFO - controller: Navigate Controller - Received Image: 16
2024-04-24 09:56:44,539 - controller - INFO - controller: Navigate Controller - command passed from child, update_setting, ('channel',)
2024-04-24 09:56:44,575 - controller - INFO - controller: Navigate Controller - Received Image: 17
2024-04-24 09:56:50,583 - controller - INFO - controller: Navigate Controller - Received Image: stop
2024-04-24 09:56:50,583 - controller - INFO - controller: Navigate Controller - Captured 218, live Images
```

AdvancedImagingUTSW, 2024-05-04:
This might be as easy as wrapping `self.ao_task.write(volts, auto_start=True)` in a `try/except` statement. Line 239, `stage_galvo.py`. Will make a pull request and try it out on one of our microscopes.

EvoleneP, 2024-05-08:
Evaluated on my machine and was able to confirm that the changes fixed the problem. I am now able to change the filter wheel in the continuous acquisition mode.

AdvancedImagingUTSW, 2024-05-08:
That's great news. I'll proceed and merge this fix.

closed 2024-05-08 by AdvancedImagingUTSW