TheDeanLab/navigate #904: Multi-Position tiling wizard does not capture current z position.

SJShep, 2024-05-24:
When using the Multi-Position tiling wizard setting the Z start/end sets the stage position to 0. It does not move the stage but changes the 

**To Reproduce**
Steps to reproduce the behavior:
1. Crop camera FOV to 2500x2500
2. Move stage to xyz = 0,0,0 
3. Set starting X,Y,Z (note that if z!=0, after setting z start, it will be changed to z=0 in the stage control gui)
    -  [image]
4. Move xyz = 2000, 2000, 500
    -  [image]
5. Set XYZ 
    - [image]
    - **NOTE:** The stage does not move to 0, it just changes the gui value. 

**Expected behavior**
Expect the tiling wizard to set z end to the current z stage position, but instead it sets it to 0.

**Traceback**
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\qi2lab\miniconda3\envs\ASInavigate\lib\tkinter\__init__.py", line 1892, in __call__
    return self.func(*args)
  File "C:\Users\qi2lab\Documents\github\asi_navigate\navigate\src\navigate\controller\sub_controllers\tiling_wizard_controller2.py", line 320, in set_table
    table_values = compute_tiles_from_bounding_box(
  File "C:\Users\qi2lab\Documents\github\asi_navigate\navigate\src\navigate\tools\multipos_table_tools.py", line 155, in compute_tiles_from_bounding_box
    thetas = dim_vector(
  File "C:\Users\qi2lab\Documents\github\asi_navigate\navigate\src\navigate\tools\multipos_table_tools.py", line 148, in dim_vector
    return start + np.arange(0, n_tiles, 1) * step
numpy.core._exceptions._UFuncNoLoopError: ufunc 'add' did not contain a loop with signature matching types (dtype('<U1'), dtype('float64')) -> None
```
[performance.log](https://github.com/TheDeanLab/navigate/files/15439112/performance.log)
[view_controller_debug.log](https://github.com/TheDeanLab/navigate/files/15439113/view_controller_debug.log)
[view_controller_info.log](https://github.com/TheDeanLab/navigate/files/15439114/view_controller_info.log)
[model_debug.log](https://github.com/TheDeanLab/navigate/files/15439115/model_debug.log)
[model_error.log](https://github.com/TheDeanLab/navigate/files/15439117/model_error.log)
[model_info.log](https://github.com/TheDeanLab/navigate/files/15439118/model_info.log)
[model_performance.log](https://github.com/TheDeanLab/navigate/files/15439119/model_performance.log)
**configuration**
```
microscopes:
  Nanoscale:
    daq:
      hardware:
        name: Dev2
        type: NI

      sample_rate: 100000
      sweep_time: 0.2

      # triggers
      master_trigger_out_line: Dev2/port0/line8
      camera_trigger_out_line: /Dev2/ctr0 # PFI12
      trigger_source: /Dev2/PFI5

      # Digital Laser Outputs
      laser_port_switcher: Dev2/port0/line0
      laser_switch_state: False

    camera:
      hardware:
        name: camera
        type: Photometrics
        camera_connection: PMPCIECam00
        serial_number: A17K631096
      subsampling: [1, 2, 4]
      readout_port: 0
      delay_percent: 5
      gain: 1
      speed_table_index: 0
      exposure_time_range:
        min: 1
        max: 1000
        step: 1
      unitforlinedelay: 10.26      

    remote_focus_device:
      hardware:
        name: daq
        type: NI
        channel: Dev2/ao2
        min: 0
        max: 10
      # FIXME: What is this? 
      # Optotune EL-16-40-TC-VIS-5D-1-C
      delay_percent: 0 #1.5 #7.5
      ramp_rising_percent: 85
      ramp_falling_percent: 1.5 #2.5
      amplitude: 0.7
      offset: 2.3
    galvo:
      -
        hardware:
          name: daq
          type: NI
          channel: Dev2/ao0
          min: 0
          max: 5
        frequency: 200
        amplitude: 1.5
        offset: 0
        duty_cycle: 50
        phase: 1.57079 # pi/2
    filter_wheel:
      hardware:
        name: filter_wheel
        type: NI  
        wheel_number: 1
      filter_wheel_delay: 0.050 # in seconds
      available_filters:
        473nm: Dev2/port0/line1
        561nm: Dev2/port0/line3
        638nm: Dev2/port0/line5
        Empty: Dev2/port0/line7
    stage:
      hardware:
        -
          name: stage
          type: MS2000
          serial_number: 1906420147517051597
          port: COM6
          baudrate: 115200
          axes: [x, y, z] # Software
          axes_mapping: [X, Y, Z] 
          feedback_alignment: [90, 90, 90, 90]
        -
          name: synthetic
          type: SyntheticStage
          serial_number: 123
          axes: [theta]
          axes_mapping: [xylophone]
          volts_per_micron: None
          axes_channels: None
          max: None
          min: None
        -
          type: KST101
          serial_number: 26001318
          axes: [f]
          axes_mapping: [1]
          feedback_alignment:
          device_units_per_mm: 2008623.6238
          volts_per_micron: 0.0
          min: 2
          max: 25
          controllername:
          stages:
          refmode:
          port:
          baudrate: 
          timeout:

      startfocus: -16000
      x_max: 10000 # Swapped from Z
      x_min: -10000 # Swapped from Z
      y_max: 10000
      y_min: -10000
      z_max: 10000 # Swapped from X
      z_min: -10000 # Swapped from X
      f_max: 25 #=m
      f_min: 0 #=m
      theta_max: 0
      theta_min: 0
      #TODO: Is it because of this external_trigger ? 
      external_trigger: /Dev2/PFI0
      # joystick_axes: [x, y, z, f]

      x_rot_position: 0
      y_rot_position: 0
      z_rot_position: 0

      x_step: 50
      y_step: 50
      z_step: 50
      theta_step: 30
      f_step: 1

      position:
        x_pos: 5 # Swapped from Z initial stage position
        y_pos: 1
        z_pos: 1 # Swapped from X
        f_pos: 0
        theta_pos: 0
      velocity: 1000

      x_offset: 0
      y_offset: 0
      z_offset: 0
      f_offset: 0
      theta_offset: 0
    
    zoom:
      hardware:
        name: zoom
        type: SyntheticZoom
        servo_id: 1
      position:
        N/A: 0
      pixel_size:
        N/A: 0.168
    shutter: 
      hardware:
        name: daq
        type: NI
        channel: Dev2/port0/line9 
        min: 0
        max: 5
    lasers:
      # Laser 473 wavelength
      - wavelength: 473
        onoff:
          hardware:
            name: daq
            type: NI
            channel: Dev2/port0/line0
            min: 0
            max: 5
        power:
          hardware:
            name: daq
            type: NI
            channel: Dev2/ao1 
            min: 0
            max: 5
        type: LuxX
        index: 0
        delay_percent: 10
        pulse_percent: 87

      # Laser 532 wavelength
      # - wavelength: 532
      #   onoff:
      #     hardware:
      #       name: daq
      #       type: NI
      #       channel: Dev2/port0/line2 
      #       min: 0
      #       max: 5
      #   power:
      #     hardware:
      #       name: daq
      #       type: NI
      #       channel: Dev2/ao2
      #       min: 0
      #       max: 5
      #   type: Obis
      #   index: 1
      #   delay_percent: 10
      #   pulse_percent: 87

      # Laser 561 wavelength
      - wavelength: 561
        onoff:
          hardware:
            name: daq
            type: NI
            channel: Dev2/port0/line4 
            min: 0
            max: 5
        power:
          hardware:
            name: daq
            type: NI
            channel: Dev2/port0/line13
            min: 0
            max: 5
        type: Obis
        index: 1
        delay_percent: 10
        pulse_percent: 87

      # Laser 638 wavelength 
      - wavelength: 638
        onoff:
          hardware:
            name: daq
            type: NI
            channel: Dev2/port0/line6
            min: 0
            max: 5
        power:
          hardware:
            name: daq
            type: NI
            channel: Dev2/ao3
            min: 0
            max: 5
        type: Obis
        index: 1
        delay_percent: 10
        pulse_percent: 87
gui:
  channels:
    count: 5
    laser_power:
      min: 0
      max: 100
      step: 10
    exposure_time:
      min: 1
      max: 1000
      step: 5
    interval_time:
      min: 0
      max: 1000
      step: 5
  stack_acquisition:
    step_size:
      min: 0.1
      max: 1000
      step: 0.1
    start_pos:
      min: -5000
      max: 5000
      step: 1
    end_pos:
      min: -5000
      max: 10000
      step: 1
  timepoint:
    timepoints:
      min: 1
      max: 1000
      step: 1
    stack_pause:
      min: 0
      max: 1000
      step: 1
```

annie-xd-wang, 2024-05-30:
In navigation, the 'Set * Start/End' function works the same way for every axis. I suspect that the stage doesn't move to the position you set in the GUI along the Z-axis. Please double-check this by moving the Z stage to a specified value and then clicking the 'Stop' button in the stage control panel. This will show where the stage device is positioned along the Z-axis. After confirming that the Z stage moves to the specified position, click 'Set Z Start/End' to set the tiling start/end position.

AdvancedImagingUTSW, 2024-05-31:
Yeah, we can do another zoom meeting to go over this